zephyrproject-rtos/zephyr #21813: tests/kernel/timer/timer_api failed on frdm_k64f board.

chen-png, 2020-01-10:
**To Reproduce**
Steps to reproduce the behavior:
1. west build -b frdm_k64f -p auto tests/kernel/timer/timer_api/
2. west flash
3. See error

**Screenshots or console output**
*** Booting Zephyr OS build zephyr-v2.1.0-803-gf05cbb421ddf ***

Running test suite timer_api
starting test - test_time_conversions

PASS - test_time_conversions
starting test - test_timer_duration_period

PASS - test_timer_duration_period
starting test - test_timer_period_0

PASS - test_timer_period_0
starting test - test_timer_expirefn_null

PASS - test_timer_expirefn_null
starting test - test_timer_periodicity

PASS - test_timer_periodicity
starting test - test_timer_status_get

PASS - test_timer_status_get
starting test - test_timer_status_get_anytime

PASS - test_timer_status_get_anytime
starting test - test_timer_status_sync

PASS - test_timer_status_sync
starting test - test_timer_k_define

**Assertion failed at ZEPHYR_BASE/tests/kernel/timer/timer_api/src/main.c:64: duration_expire: interval >= 100 is false**
FAIL - test_timer_k_define
starting test - test_timer_user_data

PASS - test_timer_user_data
starting test - test_timer_remaining_get

PASS - test_timer_remaining_get
Test suite timer_api failed.
PROJECT EXECUTION FAILED

**Environment (please complete the following information):**
 - OS: fedora28
 - Toolchain: zephyr-sdk-0.10.3
 - Commit ID: f05cbb4

dleach02, 2020-01-16:
@chen-png, I have not been able to reproduce the failure. Does this error happen to you on every run or only occasionally?

yingmingx, 2020-01-17:
I have reproduce  this failure only on the commit id f05cbb4.  The error does not happen in the later version version 5c1a3f and the commit 4e13 which is before f05cbb4.
I add 1ms sleep before the second test in the function test_timer_k_define, it passed, I think it probably have some issues on arch related porting.

@@ -412,6 +413,7 @@ void test_timer_k_define(void)
         * one will get added.
         */
        busy_wait_ms(DURATION / 2);
+       k_sleep(1);
        k_timer_start(&ktimer, DURATION, PERIOD);
        tdata.timestamp = k_uptime_get();
        busy_wait_ms(DURATION + PERIOD * EXPIRE_TIMES + PERIOD / 2);

chen-png, 2020-01-17:
it failed occasionally. 
i saw it again with commit ebaed210c, but it seems that it's difficult to reproduce.

yingmingx, 2020-01-17:
I used git bisect command and found that it was first failed at commit-id 8bdabcc.

dleach02, 2020-01-17:
Okay, I was using the tip of master to try to reproduce and was not able to have it occur.

MaureenHelm, 2020-02-24:
This test still passes on master at e4479e2feceee854f4436816c04ee98ad981b66e

closed 2020-02-24 by MaureenHelm