zephyrproject-rtos/zephyr #21848: sanitycheck duplicate tests Testing/Ztest

maksimmasalski, 2020-01-13:
**Describe the bug**
Found duplicate tests in category Testing/Ztest
```
Testing | Ztest | testing.ztest.regular_output.assert_pass | testing.ztest.regular_output.assert_pass
Testing | Ztest | testing.ztest.regular_output.assert_pass | testing.ztest.regular_output.assert_pass
Testing | Ztest | testing.ztest.regular_output.assert_pass | testing.ztest.regular_output.assert_pass
Testing | Ztest | testing.ztest.regular_output.assert_skip | testing.ztest.regular_output.assert_skip
Testing | Ztest | testing.ztest.regular_output.assert_skip | testing.ztest.regular_output.assert_skip
Testing | Ztest | testing.ztest.customized_output.assert_pass | testing.ztest.customized_output.assert_pass
Testing | Ztest | testing.ztest.customized_output.assert_pass | testing.ztest.customized_output.assert_pass
Testing | Ztest | testing.ztest.customized_output.assert_pass | testing.ztest.customized_output.assert_pass
Testing | Ztest | testing.ztest.customized_output.assert_skip | testing.ztest.customized_output.assert_skip
Testing | Ztest | testing.ztest.customized_output.assert_skip | testing.ztest.customized_output.assert_skip
```

**To Reproduce**
Steps to reproduce the behavior:
1. run $sanitycheck -export-tests testsuite.csv
2. Open using Excel.
3. Find that testcases

**Expected behavior**
All tests should have unique names.

**Impact**
Wrong count of the passed tests.

**Environment (please complete the following information):**
 - OS: Fedora 28
 - Toolchain 0.10.3
 - Latest master

nashif, 2020-01-16:
you can find duplicates more easily:
```
sanitycheck --list-test-duplicates
```

nashif, 2020-01-16:
@maksimmasalski  Please take a look at the duplicates and see if you can provide a fix...

maksimmasalski, 2020-02-13:
@nashif I submitted pr #22799, but I only solved test case names in .yaml file.
That sanitycheck command also gave me more duplicates, but that duplicates are in main.c code 
For example in code for tests I found that kind of duplicates:
`ztest_test_suite(syscalls,
ztest_unit_test(test_string_nlen),
ztest_user_unit_test(test_string_nlen),
`
I can't understand, should we use for one test case `ztest_unit_test` or `ztest_user_unit_test`
If to use them both `sanitycheck --list-test-duplicates` recognizes it like test duplicate.

nashif, 2020-03-11:
latest list:
```
Tests with duplicate identifiers:
- testing.ztest.regular_output.assert_pass
  - tests/ztest/custom_output/testing.ztest.regular_output
  - tests/ztest/custom_output/testing.ztest.regular_output
  - tests/ztest/custom_output/testing.ztest.regular_output
- testing.ztest.regular_output.assert_skip
  - tests/ztest/custom_output/testing.ztest.regular_output
  - tests/ztest/custom_output/testing.ztest.regular_output
- testing.ztest.customized_output.assert_pass
  - tests/ztest/custom_output/testing.ztest.customized_output
  - tests/ztest/custom_output/testing.ztest.customized_output
  - tests/ztest/custom_output/testing.ztest.customized_output
- testing.ztest.customized_output.assert_skip
  - tests/ztest/custom_output/testing.ztest.customized_output
  - tests/ztest/custom_output/testing.ztest.customized_output
- kernel.memory_protection.syscalls.string_nlen
  - tests/kernel/mem_protect/syscalls/kernel.memory_protection.syscalls
  - tests/kernel/mem_protect/syscalls/kernel.memory_protection.syscalls
- bluetooth.privacy.test.privacy_clear
  - tests/bluetooth/ctrl_sw_privacy_unit/bluetooth.privacy.test
  - tests/bluetooth/ctrl_sw_privacy_unit/bluetooth.privacy.test
- libraries.data_structures.byte_put_free
  - tests/lib/ringbuffer/libraries.data_structures
  - tests/lib/ringbuffer/libraries.data_structures
- usb.device.usb_dc_api_invalid
  - tests/subsys/usb/device/usb.device
  - tests/subsys/usb/device/usb.device
- net.socket.udp.v6_sendmsg_with_txtime
  - tests/net/socket/udp/net.socket.udp
  - tests/net/socket/udp/net.socket.udp
- net.util.net_addr
  - tests/net/utils/net.util
  - tests/net/utils/net.util
- net.iface.with_userspace.check_promisc_mode_off
  - tests/net/iface/net.iface.with_userspace
  - tests/net/iface/net.iface.with_userspace
- net.iface.with_userspace.netmask_addr_add
  - tests/net/iface/net.iface.with_userspace
  - tests/net/iface/net.iface.with_userspace
- net.iface.with_userspace.gw_addr_add
  - tests/net/iface/net.iface.with_userspace
  - tests/net/iface/net.iface.with_userspace
- net.iface.no_userspace_allowed.check_promisc_mode_off
  - tests/net/iface/net.iface.no_userspace_allowed
  - tests/net/iface/net.iface.no_userspace_allowed
- net.iface.no_userspace_allowed.netmask_addr_add
  - tests/net/iface/net.iface.no_userspace_allowed
  - tests/net/iface/net.iface.no_userspace_allowed
- net.iface.no_userspace_allowed.gw_addr_add
  - tests/net/iface/net.iface.no_userspace_allowed
  - tests/net/iface/net.iface.no_userspace_allowed
- net.route.populate_nbr_cache
  - tests/net/route/net.route
  - tests/net/route/net.route
- net.route.route_add
  - tests/net/route/net.route
  - tests/net/route/net.route
- net.trickle.trickle_1_status
  - tests/net/trickle/net.trickle
  - tests/net/trickle/net.trickle
i9:zephyr(master):
```

closed 2020-08-12 by github-actions[bot]