zephyrproject-rtos/zephyr #21859: Bluetooth LE Disconnect event not received

philips77, 2020-01-13:
**Describe the bug**
My fw is based on zephyr (last commit: 9a0a85cf67602e4fa5b0c6d7cef2c5e58ca4ac34). I tried zephyr LL and the old deprecated one with the same result. Phone Sony L4312 with Android 8.1 receives number of notifications with PHY LE 1M and MTU 498. After around 70 of them sent as fast as possible from the device, looks like the phone can't handle more and sends LLCP Termination. This event never gets to the app and the app is still waiting for some other command to come. Zephyr stops sending empty packets, does not advertise, just hangs there without any logs produced and a reset with a button is required. This works with PHY 2M or with MTU 23 or 247. 
[ellisys logs.zip](https://github.com/zephyrproject-rtos/zephyr/files/4053886/ellisys.logs.zip)
I tried also on other phones, but the Sony is the only one that sends this LLCP Termination. Other phones correctly receive all notifications and continue as they should. Using the Sony phone it's 100% reproducible at exactly the same moment.

**To Reproduce**
Steps to reproduce the behavior:
1. make zephyr send notifications as fast as possible to a phone/device
2. the phone/device sends LLCP Termination
3. device sends one more notification and stops responding

**Expected behavior**
The device should report disconnected event in `bt_conn_cb` and start advertising again.

**Impact**
Blocker, when tested with Sony phone

**Screenshots or console output**
```
[...]
Sending test results..................OK

Test started with parameters:
  MTU: 498
  Interval: 36 (45.00 ms)
Receiving using Write Command.........COMPLETE
  44550 bytes received in 564 ms (78.99 kB/s)
  Max packets in connection interval: 9
Sending notifications.................
```

**Environment (please complete the following information):**
 - OS: Wnidows 10
 - Toolchain (Nordic nRF Connect SDK, SES)
 - Commit 9a0a85cf67602e4fa5b0c6d7cef2c5e58ca4ac34 (zephyr folder)

**Additional context**
I'll be OOO for 2 weeks starting tomorrow.

philips77, 2020-01-13:
After 
```
Sending notifications.................
```
the log should print "Disconnected, reason...xxx".

cvinayak, 2020-01-21:
@philips77 please retry with https://github.com/zephyrproject-rtos/zephyr/pull/22005

carlescufi, 2020-02-04:
Discussed with @philips77. He will reopen if the issue reappears.

closed 2020-02-04 by carlescufi