zephyrproject-rtos/zephyr #21877: tests/drivers/uart/uart_async_api fails on qemu_cortex_m0

andrewboie, 2020-01-14:
```
To exit from QEMU enter: 'CTRL+a, x'
[QEMU] CPU: cortex-m0
*** Booting Zephyr OS build zephyr-v2.1.0-948-gebaed210c786  ***
Running test suite uart_async_test
===================================================================
starting test - test_single_read
test
    Assertion failed at ZEPHYR_BASE/tests/drivers/uart/uart_async_api/src/test_uart_async.c:59: test_single_read: (k_sem_take(&rx_rdy, K_MSEC(100)) not equal to 0)
RX_RDY timeout
FAIL - test_single_read
===================================================================
starting test - test_chained_read
Message 0
    Assertion failed at ZEPHYR_BASE/tests/drivers/uart/uart_async_api/src/test_uart_async.c:137: test_chained_read: (k_sem_take(&rx_rdy, K_MSEC(1000)) not equal to 0)
RX_RDY timeout
FAIL - test_chained_read
===================================================================
starting test - test_double_buffer

    Assertion failed at ZEPHYR_BASE/tests/drivers/uart/uart_async_api/src/test_uart_async.c:194: test_double_buffer: (uart_rx_enable(uart_dev, double_buffer[0], sizeof(double_buffer[0]), 50) not equal to 0)
Failed to enable receiving
FAIL - test_double_buffer
===================================================================
starting test - test_read_abort

    Assertion failed at ZEPHYR_BASE/tests/drivers/uart/uart_async_api/src/test_uart_async.c:248: test_read_abort: (k_sem_take(&rx_rdy, K_MSEC(100)) not equal to 0)
RX_RDY timeout
FAIL - test_read_abort
===================================================================
starting test - test_chained_write
Message 1Message 2
    Assertion failed at ZEPHYR_BASE/tests/drivers/uart/uart_async_api/src/test_uart_async.c:379: test_chained_write: (k_sem_take(&tx_done, K_MSEC(100)) not equal to 0)
TX_DONE timeout
FAIL - test_chained_write
===================================================================
starting test - test_long_buffers

    Assertion failed at ZEPHYR_BASE/tests/drivers/uart/uart_async_api/src/test_uart_async.c:453: test_long_buffers: (k_sem_take(&rx_rdy, K_MSEC(200)) not equal to 0)
RX_RDY timeout
FAIL - test_long_buffers
===================================================================
starting test - test_write_abort

    Assertion failed at ZEPHYR_BASE/tests/drivers/uart/uart_async_api/src/test_uart_async.c:309: test_write_abort: (k_sem_take(&rx_rdy, K_MSEC(100)) not equal to 0)
RX_RDY timeout
FAIL - test_write_abort
===================================================================
Test suite uart_async_test failed.
===================================================================
PROJECT EXECUTION FAILED
```

andrewboie, 2020-01-14:
For some baffling reason, this doesn't fail when sanitycheck runs it.

carlescufi, 2020-01-14:
@Mierunski any input on this issue?

Mierunski, 2020-01-14:
This test needs physical connection between RX and TX on second UART not used for console to pass

stephanosio, 2020-04-22:
This is where #21073 could potentially come in handy, given that we only have one UART on nRF51822; we could redirect console to semihosting and UART chardev to loopback.

carlescufi, 2020-05-29:
Closing this since it needs a second UART

closed 2020-05-29 by carlescufi