zephyrproject-rtos/zephyr #21914: net: dns: Answers to multiple mDNS queries sent in parallel aren't properly handled

lbise, 2020-01-14:
When trying to resolve several different mDNS names at the same time, the answers received for each queries aren't always properly matched to the initial query.

This is because mDNS queries always have an ID of 0, so when receiving an answer in dns_read, the query slot returned will always be the first of the list since all IDs are at 0:
https://github.com/zephyrproject-rtos/zephyr/blob/404aaec44dca5c2849c4e5308df30fcd7c28be99/subsys/net/lib/dns/resolve.c#L396

RFC 6762 says:
> In multicast responses, including unsolicited multicast responses, the Query Identifier MUST be set to zero on transmission, and MUST be ignored on reception.

Another way to match the answer to the query must be used for mDNS queries.

jukkar, 2020-01-15:
@lbise fix proposal at #21952. I tested it and everything seems to work ok but it would be great if you could verify that it works now as expected.

lbise, 2020-01-16:
> @lbise fix proposal at #21952. I tested it and everything seems to work ok but it would be great if you could verify that it works now as expected.

Thanks for the speedy fix jukkar. This particular issue is now fixed. I have opened another issue for another problem if you could take a look, I think the fix is trivial. https://github.com/zephyrproject-rtos/zephyr/issues/21970

PR #21952 (2020-01-24): Support multiple simultaneous mDNS requests
Fixes #21914
commit f9c18dd9 (2020-01-24): net: dns: Support multiple simultaneous mDNS requests

As mDNS requests set DNS id to 0, we cannot use it to match
the DNS response packet. In order to allow this functionality,
create a hash from query name and type, and use that together
with DNS id to match request and response.

Fixes #21914
closed 2020-01-24 by jukkar