zephyrproject-rtos/zephyr #21917: cmake error with CONFIG_COUNTER and CONFIG_BT both enabled (nrf52 board)

attdona, 2020-01-14:
cmake fails if both `CONFIG_COUNTER` and `CONFIG_BT` are set in `prj.conf`

With my actual knowledge of zephyr build system I'm not able to troubleshooting myself the problem.

**To Reproduce**
Steps to reproduce the behavior:
1. cd $ZEPHIR_HOME/samples/bluetooth/beacon
2. Add CONFIG_COUNTER=y in prj.conf
1. mkdir build; cd build
2. cmake -DBOARD=nrf52840_pca10056 ..
4. See error

```
Zephyr version: 2.1.99
-- Selected BOARD nrf52840_pca10056
-- Found west: /home/adona/dev/ZHING/zh/bin/west (found suitable version "0.6.3", minimum required is "0.6.0")
-- Loading /home/adona/dev/ZEPHYR/zephyr/boards/arm/nrf52840_pca10056/nrf52840_pca10056.dts as base
Devicetree configuration written to /home/adona/dev/ZEPHYR/zephyr/samples/bluetooth/beacon/build/zephyr/include/generated/generated_dts_board.conf
Parsing Kconfig tree in /home/adona/dev/ZEPHYR/zephyr/Kconfig
Loaded configuration '/home/adona/dev/ZEPHYR/zephyr/samples/bluetooth/beacon/build/zephyr/.config'
No change to '/home/adona/dev/ZEPHYR/zephyr/samples/bluetooth/beacon/build/zephyr/.config'
-- Cache files will be written to: /home/adona/.cache/zephyr
-- Configuring done
CMake Error at ../../../cmake/extensions.cmake:372 (add_library):
  No SOURCES given to target: drivers__counter
Call Stack (most recent call first):
  ../../../cmake/extensions.cmake:349 (zephyr_library_named)
  ../../../drivers/counter/CMakeLists.txt:3 (zephyr_library)

```

ulfalizer, 2020-01-14:
CC @SebastianBoe

SebastianBoe, 2020-01-15:
Hi, BT is using RTC0, you need to use RTC2 instead.

SebastianBoe, 2020-01-15:
e.g. in beacon/prj.conf:

```
CONFIG_BT=y
CONFIG_BT_DEBUG_LOG=y
CONFIG_BT_DEVICE_NAME="Test beacon"
CONFIG_COUNTER=y
CONFIG_COUNTER_RTC2=y

```

closed 2020-01-15 by SebastianBoe

stef, 2020-06-04:
i'm running in the same issue although on an nrf51, but here CONFIG_COUNTER_RTC2=y
it seems brings up other problems, i assume the nrf51 does not have an RTC2, can this be somehow solved though?

pabigot, 2020-06-04:
It does not.  You might try `CONFIG_COUNTER_TIMER1` or `...TIMER2`.  `TIMER0` is used by Bluetooth along with `RTC0`.

stef, 2020-06-04:
wohoo \o/ that seems to work. thanks.