zephyrproject-rtos/zephyr #21932: Current consumption on nrf52_pca10040, power_mgr sample

Mhrooczny, 2020-01-15:
Hi all,
It is my first post here and i hope that you can help me with this.

**Describe the bug**
I think that i got too high current consumption on nrf52_pca10040 devkit board. Im building `samples/boards/nrf52/power_mgr`. When board is going to `k_sleep()` function power profile shows current about ~300uA with it too much for "deep sleep".

**To Reproduce**
Steps to reproduce the behavior:
1. cd samples/boards/nrf52/power_mgr
2. west build -b nrf52_pca10040
3. west flash

**Expected behavior**
Current consumption on level ~50uA

**Impact**
I wanted to build low power project but it drains battery.

**Environment (please complete the following information):**
 - OS: Linux
 - Zephyr SDK,
 - 2.1.99

pabigot, 2020-02-17:
After some battles with PowerProfiler/PPK I can confirm this.  The 300 uA is due to the UART that is used in the sample to display the current state on the console.

I believe there is some support for UART power management, but adding these options:
```
CONFIG_DEVICE_POWER_MANAGEMENT=y
CONFIG_DEVICE_IDLE_PM=y
```

has no effect.  The minimum power can be achieved by adding these options:
```
CONFIG_CONSOLE=n
CONFIG_SERIAL=n
#CONFIG_UART_CONSOLE=n
#CONFIG_UART_NRFX=n
#CONFIG_UART_0_NRF_UART=n
```

though this produces Kconfig diagnostics that I can't get rid of, related to the commented out symbols.

The sleep power as measured by PPK with this configuration averages roughly 1.4 uA with a spike of 31 uA every 50 us, as best I can tell.

carlescufi, 2020-02-17:
@anangl @Mierunski could you comment on the analysis above by @pabigot ?

Mierunski, 2020-02-19:
With CONFIG_DEVICE_POWER_MANAGEMENT=y it is possible to manually disable UART in runtime using device_set_power_state function, calling it right before sleep will result in lower energy consumption (same goes with TWI and SPI). AFAIK there is no automatic disabling of peripherals when going to sleep.

anangl, 2020-02-19:
For UARTE, it should be possible to lower the power consumption by removing the RX pin assignment in dts overlay for this application and this way permanently deactivating RX (see https://github.com/zephyrproject-rtos/zephyr/commit/dcdc61cfdbf65b50cfdfa01e7f40daf08a58615a). There should be no need to call `device_set_power_state` then. For UART, a corresponding patch is not yet available, but it is supposed to appear, hopefully soon.

Mhrooczny, 2020-02-19:
Thank you for reply. How to make it work using `device_set_power_state`? As i understand it is needed to make UART as a device like in samples with led? I mean `device_get_binding` i tried to make additional alias for uart in .dts like this:
```
alliases{
uart0 = &uart0;
}
```
then define this uart in main:
`#define UART0 DT_ALIAS_UART0`
then:
`struct device *dev = device_get_binding(UART0);`

But get only `error: 'DT_ALIAS_UART0' undeclared here (not in a function); did you mean 'DT_ALIAS_UART0_IRQ'?`

I'm sorry if it is obvious but i'm newbie to Zephyr.

Mierunski, 2020-02-21:
You need to pass device label to device_get_binding function, i.e. `DT_UART_0_NAME` or `DT_ALIAS_UART_0_LABEL`. You don't need overlays for it, to check generated defines look in build directory `/build/zephyr/include/generated/` for files devicetree_unfixed.h and devicetree_fixups.h

Mhrooczny, 2020-02-24:
When i pass `DT_UART_0_NAME` or `DT_ALIAS_UART_0_LABEL` to `device_get_binding` function im getting error: `error: initializer element is not constant`. I don't have files like devicetree_unfixed.h or devicetree_fixups.h but just generated_dts_board_fixups.h and generated_dts_board_unfixed.h but i think that this is same thing.

PR #25490 (2020-05-25): samples: nrf: system_off: demonstrate use of device pm to reduce power
Currently system power management does not automatically control device power, and this example demonstrates high current draw due to the UART being enabled.  Enable device power management and demonstrate how to manually turn off greedy devices.
    
Also remove the 60 s delay that confirms we successfully turned off the residency policy; that's not directly relevant to the purpose of the sample.

Fixes #21932 albeit for `system_off` not `power_mgr` which was removed because it misused API.
closed 2020-05-25 by carlescufi