zephyrproject-rtos/zephyr #21955: usb: tests/subsys/usb/device fails on all NXP RT boards

hakehuang, 2020-01-15:
**Describe the bug**
see log

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=mimxrt1050_evk ..
cmake -DBOARD=mimxrt1015_evk ..
cmake -DBOARD=mimxrt1060_evk ..
cmake -DBOARD=mimxrt1064_evk ..
cmake -DBOARD=mimxrt1020_evk ..
3. make
4. make flash
5 see the error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
usb device function support

**Screenshots or console output**
```
*** Booting Zephyr OS version 2.1.99  ***

D: lock usb_enable_lock mutex

D: attached

D: set cb, ep: 0x81

D: unlock usb_enable_lock mutex

Running test suite test_device

===================================================================

starting test - test_usb_dc_api_invalid

    Assertion failed at ZEPHYR_BASE/tests/subsys/usb/device/src/main.c:124: test_usb_dc_api_invalid: (usb_dc_ep_set_stall(INVALID_EP) equal to TC_PASS)

usb_dc_ep_set_stall(INVALID_EP)

FAIL - test_usb_dc_api_invalid

===================================================================

starting test - test_usb_dc_api

PASS - test_usb_dc_api

===================================================================

starting test - test_usb_dc_api_read_write

E: Wrong endpoint index/address/direction

    Assertion failed at ZEPHYR_BASE/tests/subsys/usb/device/src/main.c:187: test_usb_dc_api_read_write: (usb_write(INVALID_EP, &byte, sizeof(byte), &size) equal to TC_PASS)

usb_write(INVALID_EP)

FAIL - test_usb_dc_api_read_write

===================================================================

starting test - test_usb_dc_api_invalid

    Assertion failed at ZEPHYR_BASE/tests/subsys/usb/device/src/main.c:124: test_usb_dc_api_invalid: (usb_dc_ep_set_stall(INVALID_EP) equal to TC_PASS)

usb_dc_ep_set_stall(INVALID_EP)

FAIL - test_usb_dc_api_invalid

===================================================================

starting test - test_usb_deconfig

PASS - test_usb_deconfig

===================================================================

starting test - test_usb_disable

D: detached.

PASS - test_usb_disable

===================================================================

Test suite test_device failed.

===================================================================

PROJECT EXECUTION FAILED

**Environment (please complete the following information):**
 - OS: (e.g. Linux)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit: 2.1.99

hakehuang, 2020-01-15:
@MaureenHelm

jfischer-phytec-iot, 2020-01-16:
#21175 should fixes this issue

jfischer-phytec-iot, 2020-02-05:
#19763 is the same issue

carlescufi, 2020-05-13:
Fixed

closed 2020-05-13 by carlescufi