zephyrproject-rtos/zephyr #21962: drivers: usb: usb_dc_stm32: does not compile for stm32f3_disco board

rckTom, 2020-01-15:
**Describe the bug**
tests/subsys/usb/device (specifically drivers/usb/device/usb_dc_stm32.c) does not compile for stm32f3_disco board. Bisecting the problem leads to commit 7165803 which introduced a new version of the STM32Cube packages.

**To Reproduce**
Steps to reproduce the behavior:
1. cd test/usb/subsys/device
2. west build -b stm32f3_disco

**Expected behavior**
USB driver should compile for stm32f3_disco board as it has with older versions of STM32Cube.

galak, 2020-01-15:
I think this might be the fix, but need @erwango or someone else more familiar with the usb & stm32 code to verify:

```
diff --git a/drivers/usb/device/Kconfig b/drivers/usb/device/Kconfig
index 7bca8ccc22..40e5a8f695 100644
--- a/drivers/usb/device/Kconfig
+++ b/drivers/usb/device/Kconfig
@@ -30,7 +30,7 @@ config USB_DC_STM32
        bool "USB device controller driver for STM32 devices"
        depends on SOC_FAMILY_STM32
        select USB_DEVICE_DRIVER
-       select USE_STM32_LL_USB if !SOC_SERIES_STM32F0X && !SOC_SERIES_STM32F3X && !SOC_SERIES_STM32L0X
+       select USE_STM32_LL_USB
        select USE_STM32_HAL_PCD
        select USE_STM32_HAL_PCD_EX
        help
diff --git a/drivers/usb/device/usb_dc_stm32.c b/drivers/usb/device/usb_dc_stm32.c
index eb232422da..4e337743d9 100644
--- a/drivers/usb/device/usb_dc_stm32.c
+++ b/drivers/usb/device/usb_dc_stm32.c
@@ -59,20 +59,6 @@ LOG_MODULE_REGISTER(usb_dc_stm32);
 #error "Only one interface should be enabled at a time, OTG FS or OTG HS"
 #endif
 
-/*
- * USB LL API provides the EP_TYPE_* defines. STM32Cube does not
- * provide USB LL API for STM32F0, STM32F3 and STM32L0 families.
- * Map EP_TYPE_* defines to PCD_EP_TYPE_* defines
- */
-#if defined(CONFIG_SOC_SERIES_STM32F3X) || \
-       defined(CONFIG_SOC_SERIES_STM32F0X) || \
-       defined(CONFIG_SOC_SERIES_STM32L0X)
-#define EP_TYPE_CTRL PCD_EP_TYPE_CTRL
-#define EP_TYPE_ISOC PCD_EP_TYPE_ISOC
-#define EP_TYPE_BULK PCD_EP_TYPE_BULK
-#define EP_TYPE_INTR PCD_EP_TYPE_INTR
-#endif
-
 /*
  * USB and USB_OTG_FS are defined in STM32Cube HAL and allows to distinguish
  * between two kind of USB DC. STM32 F0, F3, L0 and G4 series support USB device
```

ydamigos, 2020-01-16:
@galak Your patch seems reasonable.

PR #21969 (2020-01-16): drivers/usb/device: stm32: Update due to API change on F0/F3/L0
Following update of STM32Cube packages for series F0/L0/F3,
a new file ll_usb.h is now available for these series.
As a consequence, specific hanlding is no more requested for this
series is stm32 usb_device driver.

Fixes #21962
commit 136d92de (2020-01-16): drivers/usb/device: stm32: Update due to API change on F0/F3/L0

Following update of STM32Cube packages for series F0/L0/F3,
a new file ll_usb.h is now available for these series.
As a consequence, specific hanlding is no more requested for this
series is stm32 usb_device driver.

Fixes #21962
commit feb9ef10 (2020-01-16): drivers/usb/device: stm32: Update due to API change on F0/F3/L0

Following update of STM32Cube packages for series F0/L0/F3,
a new file ll_usb.h is now available for these series.
As a consequence, specific hanlding is no more requested for this
series is stm32 usb_device driver.

Fixes #21962
closed 2020-01-16 by nashif