zephyrproject-rtos/zephyr #21967: json: json_obj_parse will modify the input string

codingspirit, 2020-01-16:
**Describe the bug**
The end `"` of string items will be changed to `\0` after `json_obj_parse`, i.e.  we have a char array like this:
```
{
    "method": "hello",
    "id": 1,
}
```
if using `json_obj_parse` to proceed it, then it will be changed to 
```
{
    "method": "hello  ?, //here " symbol changed to "\0"
    "id": 1,
}
```

**To Reproduce**
Steps to reproduce the behavior:
1. create a json char array with string items
2. decode it with `json_obj_parse`
3. print out the json char array
4. see modifications

**Expected behavior**
From my point of view, `json_obj_parse` should be
```
int json_obj_parse(const char *payload, size_t len,
		   const struct json_obj_descr *descr, size_t descr_len,
		   void *val)
```
which means the input json string should not be modified.

**Impact**
Sometimes we'd like to proceed a json string twice:
1. the first time only proceed the first filed to know the info of this json
2. according the info we parse again with specific `json_obj_descr`
If `json_obj_parse` will modify string, user have to create a copy of string .

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK
 - Commit SHA or Version used: f05cbb421ddfea03961b552977e4336d0ebdef63

mbolivar, 2020-01-28:
Is this really a bug? I think this is the library working as designed. Otherwise it would have to duplicate the string.

mbolivar, 2020-02-12:
@jhedberg I think the labels need a -bug +question and this issue can be closed.

jhedberg, 2020-02-12:
> (Not sure why I was assigned)

@dcpleung unfortunately I can't remember, but seems like a mistake - sorry about that.

> @jhedberg I think the labels need a -bug +question and this issue can be closed.

@mbolivar agreed.

closed 2020-02-12 by jhedberg