zephyrproject-rtos/zephyr #21974: make include hierarchy consistent with expected usage

pabigot, 2020-01-16:
In response to a review comment I proposed #21776 which removed the implicit addition of `${ZEPHYR_BASE}/include/drivers` from the default include path, so that `<drivers/sensor.h>` would be enforced.  This breaks bluetooth, which wants to use `<bluetooth/whatever.h>` but the location is `${ZEPHYR_BASE}/include/drivers/bluetooth`.

It would be nice if the two were consistent, which probably means `bluetooth` should move up one level.

SebastianBoe, 2020-01-16:
FYI: @jhedberg , @joerchan , @cvinayak

joerchan, 2020-01-16:
In `include/drivers/bluetooth` there is only `hci_driver.h`. Most HCI drivers are implemented in Zephyr itself. So only out-of-tree HCI drivers would be affected by this. 

I don't mind switching to `drivers/bluetooth/hci_driver.h` to make in consistent. It would require updating our downstream HCI driver but shouldn't pose any big issues.
I`m currently not aware of any other maintaining their hci driver out-of-tree though.

@aescolar Any comments?

joerchan, 2020-01-16:
Actually it appears both paths are used. And our downstream use `drivers/bluetooth/hci_driver.h` so only zephyr HCI drivers are affected (that I'm aware of).

PR #21978 (2020-01-17): Bluetooth: Always use full path to hci_driver.h
The path to include/drivers should not be in the compiler include path
list, only include/. In order to make this possible, always explictly
refer to hci_driver.h via the drivers/bluetooth/ path and not only
bluetooth/.

Fixes #21974.
commit 6cf7ac77 (2020-01-17): Bluetooth: Always use full path to hci_driver.h

The path to include/drivers should not be in the compiler include path
list, only include/. In order to make this possible, always explictly
refer to hci_driver.h via the drivers/bluetooth/ path and not only
bluetooth/.

Fixes #21974.
closed 2020-01-17 by nashif