zephyrproject-rtos/zephyr #21980: Doesnt Install on Raspberry Pi

tgwaste, 2020-01-16:
During:
pip3 install --user -r ~/zephyrproject/zephyr/scripts/requirements.txt

There were lots of errors with regard to **No package 'libffi' found**
So I installed: **apt-get -y install python-cffi libffi-dev**

```
Now The error is:

    Complete output from command python setup.py egg_info:
    c/_cffi_backend.c:2:10: fatal error: Python.h: No such file or directory
     #include <Python.h>
              ^~~~~~~~~~
    compilation terminated.
    Traceback (most recent call last):
      File "/usr/lib/python3.7/distutils/unixccompiler.py", line 118, in _compile
        extra_postargs)
      File "/usr/lib/python3.7/distutils/ccompiler.py", line 909, in spawn
        spawn(cmd, dry_run=self.dry_run)
      File "/usr/lib/python3.7/distutils/spawn.py", line 36, in spawn
        _spawn_posix(cmd, search_path, dry_run=dry_run)
      File "/usr/lib/python3.7/distutils/spawn.py", line 159, in _spawn_posix
        % (cmd, exit_status))
    distutils.errors.DistutilsExecError: command 'arm-linux-gnueabihf-gcc' failed with exit status 1

    During handling of the above exception, another exception occurred:
```

Would be awesome if this worked on Pi.

nashif, 2020-01-16:
install python development packages

tgwaste, 2020-01-16:
Please be more specific.

carlescufi, 2020-01-17:
Marked as enhancement since we do not officially support development on the Raspberry Pi.

carlescufi, 2020-01-17:
@tgwaste 

> Please be more specific. Python is already nightmarish enough.

`sudo apt-get install python-dev`

tgwaste, 2020-01-17:
Thats already installed so cant be the issue:
**python-dev is already the newest version (2.7.16-1).**

aunsbjerg, 2021-01-21:
> Thats already installed so cant be the issue:
> **python-dev is already the newest version (2.7.16-1).**

Try with `python3-dev`. Looks like your installing python 2.7 dev packages but using pip3

gmarull, 2022-08-11:
I think this has been already answered, and RPi is not an officially supported platform, so closing.

closed 2022-08-11 by gmarull