zephyrproject-rtos/zephyr #21984: i2c_4 not working on stm32f746g_disco

DenvE, 2020-01-16:
**Describe the bug**
Enabling I2C_4 on stm32f746g_disco is not working as the pinmux doesn't have the pin mapping for I2C4 ( exp STM32F7_PINMUX_FUNC_PD13_I2C4_SDA)

**To Reproduce**
1. add to proj.conf
# Enable I2C4
CONFIG_I2C=y
CONFIG_I2C_4=y

CONFIG_HAS_DTS_I2C=y

CONFIG_I2C_STM32_INTERRUPT=y
CONFIG_I2C_INIT_PRIORITY=60

2. enable i2c4 in board dts
&i2c4 {
	status = "okay";
	clock-frequency = <I2C_BITRATE_FAST>;
};

3. add in pinmux.c mapping of i2c4
#ifdef CONFIG_I2C_4
	{ STM32_PIN_PD12, STM32F7_PINMUX_FUNC_PD12_I2C4_SCL },
	{ STM32_PIN_PD13, STM32F7_PINMUX_FUNC_PD13_I2C4_SDA },
#endif
3. build you application
4. error STM32F7_PINMUX_FUNC_PD13_I2C4_SDA not defined

**Expected behavior**
According data sheet stm32f7 i2c4 can be mapped on different gpio, these mapping should be define.

**Impact**
Can not use i2c4

**Environment (please complete the following information):**
 - OS:  MacOS
 - Toolchain ( Zephyr  r 2.1.99)

erwango, 2020-01-17:
This is not a bug. 
If I2C4 is not configured for this board or the series, then support should be added, this is an enhancement.

DenvE, 2020-01-17:
Hi Erwan,

sorry for this, I have done already a pull request.

i2c4 was not enabled on disco board in zephyr, but I have an own board with
this controller and have added i2c4. It was missing these pinmux defines in
the zephyr tree.  That is why I though it was a bug.

To be clear: if not yet added to supported boards, it should consider as
enhancement.

Kr,

Wim

erwango, 2020-05-11:
@DenvE Do you still need this ?

DenvE, 2020-05-12:
Hi Erwan,

I do still need this.  I'm quite busy with other project and could not
followup the status

kr,

Wim

erwango, 2020-05-25:
@denvE, there is no plan to push this change on or side, but we'll be pleased to assist you in merging this change if you want to push it. Hence, I'm closing current issue.

closed 2020-05-25 by erwango