zephyrproject-rtos/zephyr #21989: websocket: recv_msg always returns full message length on last call

mrfuchs, 2020-01-16:
`websocket_recv_msg()` always returns full message length on last call:
https://github.com/zephyrproject-rtos/zephyr/blob/ba98e72db2f0c921ebb6a6d957a9c3609daec354/subsys/net/lib/websocket/websocket.c#L821

This is probably meant to limit the length to the actual message length when the buffer contains additional trailing data (of another message) and is fine when messages are entirely read at once.

However, using the MQTT client with `transport.type = MQTT_TRANSPORT_SECURE_WEBSOCKET` for example, when reading publish payload data in chunks as it is done in
https://github.com/zephyrproject-rtos/zephyr/blob/ba98e72db2f0c921ebb6a6d957a9c3609daec354/samples/net/cloud/google_iot_mqtt/src/protocol.c#L148-L151
then the `remaining_payload` variable will underflow in this line:
https://github.com/zephyrproject-rtos/zephyr/blob/ba98e72db2f0c921ebb6a6d957a9c3609daec354/subsys/net/lib/mqtt/mqtt.c#L661

Limiting `recv_len` by `recv_len = MIN(recv_len, ctx->message_len);` works for me.

mrfuchs, 2020-01-16:
> Limiting `recv_len` by `recv_len = MIN(recv_len, ctx->message_len);` works for me.

Looking at this once more, though it worked for me, this fix obviously is not right. It should fail when payload data is read in chunks and the input buffer has trailing data.

jukkar, 2020-01-17:
Can you try if this works in your setup
```
diff --git a/subsys/net/lib/websocket/websocket.c b/subsys/net/lib/websocket/websocket.c
index a066e2190f..5bb58d2752 100644
--- a/subsys/net/lib/websocket/websocket.c
+++ b/subsys/net/lib/websocket/websocket.c
@@ -818,7 +818,6 @@ int websocket_recv_msg(int ws_sock, u8_t *buf, size_t buf_len,
                        ctx->tmp_buf_len = 0;
                }
 
-               recv_len = ctx->message_len;
                ctx->header_received = false;
 
                if (remaining) {
```
It looks like we should not modify recv_len here.

mrfuchs, 2020-01-17:
Thanks, @jukkar. Unfortunately, this fails as well. It looks like `recv_len` includes the 2 bytes of the  websocket header then (e.g. `message_len == 4`, `header_len == 2`, `recv_len == 6`), but `websocket_recv_msg` is expected to return the message length only.

jukkar, 2020-01-18:
> Unfortunately, this fails as well. It looks like `recv_len` includes the 2 bytes of the websocket header then (e.g. `message_len == 4`, `header_len == 2`, `recv_len == 6`), but `websocket_recv_msg` is expected to return the message length only.

Indeed. Thanks for the verification, I will try to figure out a better way to do this.

mrfuchs, 2020-01-20:
Thanks. Could you please also take a look at line 731?
https://github.com/zephyrproject-rtos/zephyr/blob/ba98e72db2f0c921ebb6a6d957a9c3609daec354/subsys/net/lib/websocket/websocket.c#L731
Isn't this redundant to what we are doing in line 702, because there's no more assignment to `recv_len` in between these lines?
https://github.com/zephyrproject-rtos/zephyr/blob/ba98e72db2f0c921ebb6a6d957a9c3609daec354/subsys/net/lib/websocket/websocket.c#L702

jukkar, 2020-01-21:
> Isn't this redundant to what we are doing in line 702, because there's no more assignment to `recv_len` in between these lines?

Yes, it really looks like that line 702 addition should not be there. I try to add some test cases for these. Testing this beast is a bit tricky as I do not want to end up implementing a websocket server.

jukkar, 2020-01-21:
@mrfuchs I have a fix proposal at #22064. I have created unit test for the websocket RX and it seems to work ok. Could you try it and confirm it works ok for your setup too?

mrfuchs, 2020-01-22:
Thanks, @jukkar. Unfortunately, it doesn't work for me. If the message is received as a whole, `header` holds complete websocket message, `tmp_buf` holds the payload (without the header), but `buf` is untouched (i.e. all zero) and still `recv_len == header_len + message_len`.

jukkar, 2020-01-22:
> Thanks, @jukkar. Unfortunately, it doesn't work for me. If the message is received as a whole

Thanks for the info! After adding a test for this (reading whole message in one go), the unit test fails too. I will figure out the culprit.

jukkar, 2020-01-24:
@mrfuchs New fixed version submitted to the PR. It has now more tests and I also verified against `zephyr-websocket-server.py` that is found in net-tools, that the websocket sample application works as expected. Hopefully this works in your setup, please try and report any issues.

mrfuchs, 2020-01-24:
Works for me!

PR #22064 (2020-01-28): Fix websocket when reading small amount of data
Also add unit test for that.

Fixes #21989
commit cb754786 (2020-01-28): net: websocket: Fix receiving websocket data

Make sure that if we receive websocket data in small chunks,
the parsing and returning of data to caller is done properly.

Fixes #21989
closed 2020-01-28 by jukkar