zephyrproject-rtos/zephyr #21991: memory domain locking may not be entirely correct

andrewboie, 2020-01-16:
**Describe the bug**
The memory domain APIs have a global spinlock in kernel/mem_domain.c to serialize access to memory domain data structures. However, the memory domain structures may also be accessed from within the arch code, without holding these locks.

For example, on x86, arch_user_mode_enter() examines the thread's current memory domain object and updates page tables based on its contents, without any lock held. It's possible that the thread's memory domain may be simultaneously manipulated by another thread.

On ARM, which is uniprocessor-only, the memory domain is accessed in the context switch code to reprogram the MPU. This happens with IRQs locked, so I think we are safe for now (at least until we have some sort of ARM SMP support).

We need the locking to protect:
1. The struct k_mem_domain itself
2. For any thread that is a member of the memory domain, its assignment and membership dlist.

A semaphore may suffice, a spinlock may not be required. I don't think we have use-cases for memory domain APIs being called from ISRs.

**To Reproduce**
Found via code inspection.

**Impact**
It's possible that the thread could obtain an undefined memory domain configuration. Depending on what the arch code does with the data structures, this might also crash.

andrewboie, 2020-01-16:
Leaving this as low priority for now until we can do some deeper analysis, but wanted to get this in GH.

ioannisg, 2020-01-31:
>On ARM, which is uniprocessor-only, the memory domain is accessed in the context switch code to reprogram the MPU. This happens with IRQs locked, so I think we are safe for now (at least until we have some sort of ARM SMP support).

@andrewboie I am not sure if you are referring to the re-programming of memory domains in ARM context- switch. If yes, then this statement is not correct; there is no locking in context-switch when we re-program the domains.

ioannisg, 2020-01-31:
>  I don't think we have use-cases for memory domain APIs being called from ISRs. 

If this is the case @andrewboie could we enforce this one API-level? I'd be very happy to do that :)

dleach02, 2020-07-20:
Let me also add that the memory domain MPU mapping on ARM seems a bit pointless. My limited review of the ARM MPU initialization is that implementations appear to create a static area that covers all of RAM. When the threads are swapped in the z_arm_configure_dynamic_mpu_regions() will create a MPU map that includes the memory domain(s) associated with the thread, the thread stack, and the stack guard band (if enabled). But the memory domain(s) were already mapped in by the static initialization of the MPU at startup.

I'm not sure how the x86 handles this, but it seems like when a thread swap occurs, you really want to have a complete remapping of the dynamic RAM so that user mode thread can only access the memory areas it is approved to access. That would imply that any memory domains that belong to other threads should not be mapped into the current thread's space. But the current thread swap logic doesn't seem to have any view of this given how we are currently remapping the memory space. Is it possible that the configuration of the dynamic MPU regions is supposed to remove the mapping done at init for all of memory and the thread will only have access to memory domains and stack?

andrewboie, 2020-07-20:
> But the memory domain(s) were already mapped in by the static initialization of the MPU at startup.

@dleach02 I don't believe that's true. The ARM code is a little confusing because `k_mem_partition` is over-loaded and uses for purposes outside of memory domains. I would actually like to see this changed and CONFIG_MEMORY_PROTECTION deleted so that 1) the ARM MPU code doesn't depend on the `k_mem_partition` defintion even when USERSPACE is disabled 2) I can change memory domain structures without worrying about breaking the ARM MPU code. (cc @ioannisg)

At any rate, there are some MPU regions reserved, at boot, for static regions like program text, rodata. These regions don't change on context switch, everyone can access them.

Memory domains are concerned with granting user threads access to memory beyond that -- A user thread cannot otherwise access any ram besides their own stack and whatever they were granted at boot.

I do believe we have tests that prove this works.

If you are seeing a situation where user threads are gaining access to memory they shouldn't please do let me know the specifics.

andrewboie, 2020-07-20:
> If this is the case @andrewboie could we enforce this on API-level? I'd be very happy to do that :)

@ioannisg Sure we can add this. Feel free to send a patch.

In the fullness of time, I want to redesign the memory domain APIs later this year / early next year, after virtual memory lands.

dleach02, 2020-07-21:
part of my comment turns out to be my confusion when debugging the test case between a supervisor thread and a user thread. A supervisor thread will end up having dynamic memory setup with a partition for memory domain with attributes for user and supervisor modes. It will then create a hole for the stack guard when for the NXP mpu code the hole will map from beginning of memory to the start of the guard then end of guard to end of memory. The MPU descriptors for these two memory blocks will only be configured for supervisor mode rw- with the hole having r-x

PR #28986 (2020-10-20): userspace: address some memory domain issues
The PR fixes some issues:

- When threads exit, they were not being removed from the list of threads in the memory domain they belonged to. This wasn't causing major issues yet, but will when we need to adjust shared page tables on thread exit.
- The spinlock for memory domains is no longer static and the x86 code was adjusted to hold it when a thread drops to user mode and needs to update its page tables
- `k_mem_domain_destroy()` no longer leaves threads without a memory domain assignment

Also:
- Remove an unnecessary NULL check in the x86 code
- [... cut]
closed 2020-10-20 by andrewboie