zephyrproject-rtos/zephyr #21994: Bluetooth: controller: split: Fix procedure complete event generation

cvinayak, 2020-01-17:
**Describe the bug**
Generate PHY update complete and Conn Update events after the on-air instant

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=board\_xyz
3. make
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

showstopper

**Screenshots or console output**
If applicable, add a screenshot (drag-and-drop an image), or console logs
(cut-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue.

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

**Additional context**
Add any other context about the problem here.

cvinayak, 2020-01-17:
Use inspiration from legacy controller: zephyrproject-rtos/zephyr@2bf9d49

carlescufi, 2020-02-18:
@kruithofa @cvinayak is this still an issue on the current EBQ runs?

cvinayak, 2020-02-20:
@carlescufi This is not an issue for the runs, we have workaround.

cvinayak, 2020-05-22:
@carlescufi this depends on new LLCP work, not worth new effort to fix this in current split controller.

PR #44791 (2022-05-21): Bluetooth: Controller: Fix instant based procedure complete event
Fix instant based procedure complete event generation to be
held until after the on-air instant has elapsed.

Fixes #21994.
closed 2022-05-21 by mbolivar-nordic