zephyrproject-rtos/zephyr #21996: Native POSIX or QEMU X86 emulation does not detect Bluetooth HCI Vendor-Specific Extensions

leewkb4567, 2020-01-17:
**Describe the bug**
When running on Native POSIX or QEMU X86, Bluetooth peripheral examples does not use the static random address in PCA10040 (nRF52 DK). Instead the Bluetooth device address is randomly generated. It is found that Bluetooth HCI Vendor-Specific Extensions is not enabled in these build targets.

**To Reproduce**
1. Connect a PCA10040 running hci_uart sample to Linux system.
2. Attach hci_uart to Linux Bluetooth.
    sudo btattach -B /dev/ttyACM0 -S 1000000 -R
3. Build peripheral_dis sample for board native_posix.
    west build -b native_posix -d build/native_posix
4. Run peripheral_dis sample.
   sudo build/native_posix/zephyr/zephyr.exe -bt-dev=hci0
5. Check device Bluetooth address in nRF Connect. It is not the same as the address by a nRF5 SDK BLE example.
6. Abort Zephyr executable by Ctrl-C. Run the sample again. nRF Connect will show the device using a different Bluetooth address.
7. QEMM X86 board target shows the same scenario. The steps needs to be adjusted. btproxy needs to be run.

**Expected behavior**
The Bluetooth peripheral example should be able to use the static random address in nRF52 SoC.

**Environment**
 - OS: Debian 9.5 X64 running in VirtualBox 5.2
 - Toolchain: Zephyr SDK 0.10.3
 - Zephyr v2.1.0. Issue still applies to latest Commit SHA bb32e9e3fbad4570c90f02220426b69bb9182ed0

**Additional context**
Kconfig option BT_HCI_VS_EXT_DETECT is supposed to detect Zephyr HCI VS extensions. Nevertheless, this option is disabled in POSIX and QEMU targets because HCI VS is not enabled.

The Bluetooth controllers signal the HCI VS presence by following line:
	select BT_HAS_HCI_VS
POSIX and QEMU targets do not have BT_HAS_HCI_VS enabled and thus no HCI VS commands.

Applying the following hack, Bluetooth peripheral example can use the static random address by detecting the HCI VS extensions.
$zephyr\subsys\bluetooth\common\Kconfig
...
config BT_HAS_HCI_VS
	bool
\+	default y if BOARD_QEMU_X86 || BOARD_QEMU_CORTEX_M3 || BOARD_NATIVE_POSIX
	help
...

jhedberg, 2020-01-17:
It's not safe to have the vendor extensions enabled by default when we don't know for sure what controller is connected. HCI vendor commands all share the same opcodes, so it's possible that they map to some completely different vendor command set, and could cause harmful behaviour. It's therefore best to leave the extensions disabled by default and leave it up to the discretion of the user to enable them when they know for sure that the connected controller supports the extensions or are willing to take the risk of incorrect identification (since there's no 100% runtime-reliable way).

One improvement we could perhaps do (if it's not already the case) is to document this better so users can more easily discover this information. Rather than doing the changes you've done to the board files I'd suggest to modify the configuration of your application instead.

jhedberg, 2020-01-17:
FYI @joerchan @carlescufi, @Vudentz in case you have any opinion on this

joerchan, 2020-01-17:
@jhedberg There seems to be a Kconfig issue here. Because the BT_HCI_VS cannot be selected unless BT_HAS_HCI_VS is set, which is a hidden option used by the controllers to claim Zephyr HCI VS Support. This option cannot be set by the application AFAIK.

At the moment BT_HCI_VS_EXT_DETECT cannot be set for NATIVE_POSIX, even though it is default to yes for this board.

jhedberg, 2020-01-17:
@joerchan good point - I didn't realise this. Perhaps the `depends on` should be removed and the default be made `default BT_HAS_HCI_VS`?

joerchan, 2020-01-17:
@leewkb4567 With the given fix. It should work by adding these options to your configuration.
BT_HCI_VS
BT_HCI_VS_EXT

(BT_HCI_VS_EXT_DETECT should be set to default y in this case).

PR #22001 (2020-01-17): Bluetooth: HCI: Fix allow Zephyr VS HCI commands in host-only build
Fix unable to select the Zephyr Vendor-Specific HCI commands support in
host-only build. Set VS HCI support as default on if it is known that
the controller supports it. Otherwise set it to off, this means that
VS HCI support will be default off in host-only builds.
Change default value of VS HCI detection to always on in host-only build
since we cannot be sure of VS HCI support in this case. This option is
only available in host-only builds when VS HCI has been enabled.

Fixes: #21996
commit 8b98231e (2020-01-17): Bluetooth: HCI: Fix allow Zephyr VS HCI commands in host-only build

Fix unable to select the Zephyr Vendor-Specific HCI commands support in
host-only build. Set VS HCI support as default on if it is known that
the controller supports it. Otherwise set it to off, this means that
VS HCI support will be default off in host-only builds.

Fixes: #21996
closed 2020-01-17 by jhedberg