zephyrproject-rtos/zephyr #22027: [00:00:00.051,000] <dbg> i2c_ll_stm32_v1.msg_end: stm32_i2c_msg_write: NACK

boonpin1975, 2020-01-18:
building the i2c_scanner on olimexino_stm32 , with the sht3xd sensor connected to i2c1 , I keep getting this error when running i2c_scanner 

[00:00:00.051,000] <dbg> i2c_ll_stm32_v1.msg_end: stm32_i2c_msg_write: NACK 

is there need extra kernel config when using the i2c API in stm32  ? 

Starting i2c scanner...

     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         

0 devices found on I2C_1
prj.conf
CONFIG_I2C=y
CONFIG_LOG=y
CONFIG_STDOUT_CONSOLE=y
CONFIG_PRINTK=y
CONFIG_I2C_LOG_LEVEL_DBG=y

boonpin1975, 2020-01-20:
I have tested another eeprom , at24c02 , address is 0x50 , but there is nothing in the i2c_scanner console .

erwango, 2020-01-20:
Cf #20325:
"
[...]  the test doesn't work because the ST I2C driver doesn't handle the SMBUS command that's used to detect a NAK on the address for an absent subordinate device. It reports all addresses as being associated with devices."

Moving this to enhancement.

pabigot, 2020-08-25:
#20560 fixed the I2C scanner on STM32, so if this is reproducible there's something else going on.  Possibly scanning the wrong I2C bus.

nashif, 2021-05-28:
please file a bug if this still occurs.

closed 2021-05-28 by nashif