zephyrproject-rtos/zephyr #22036: sanitycheck for qemu_cortex_r5 fails

andrewboie, 2020-01-20:
```
Deleting output directory /dev/shm/sanity-out
INFO    - JOBS: 48
INFO    - Building initial testcase list...
INFO    - 87 test configurations selected, 647 configurations discarded due to filters.
INFO    - Adding tasks to the queue...
INFO    - Total complete:   24/  87  27%  skipped:   22, failed:    0
ERROR   - qemu_cortex_r5            samples/philosophers/sample.kernel.philosopher.semaphores FAILED: Exited with 2
ERROR   - see: /dev/shm/sanity-out/qemu_cortex_r5/samples/philosophers/sample.kernel.philosopher.semaphores/build.log
INFO    - Total complete:   25/  87  28%  skipped:   22, failed:    1
ERROR   - qemu_cortex_r5            samples/philosophers/sample.kernel.philosopher     FAILED: Exited with 2
ERROR   - see: /dev/shm/sanity-out/qemu_cortex_r5/samples/philosophers/sample.kernel.philosopher/build.log
INFO    - Total complete:   26/  87  29%  skipped:   22, failed:    2
ERROR   - qemu_cortex_r5            samples/philosophers/sample.kernel.philosopher.same_prio FAILED: Exited with 2
ERROR   - see: /dev/shm/sanity-out/qemu_cortex_r5/samples/philosophers/sample.kernel.philosopher.same_prio/build.log
INFO    - Total complete:   27/  87  31%  skipped:   22, failed:    3
ERROR   - qemu_cortex_r5            samples/philosophers/sample.kernel.philosopher.static FAILED: Exited with 2
ERROR   - see: /dev/shm/sanity-out/qemu_cortex_r5/samples/philosophers/sample.kernel.philosopher.static/build.log
INFO    - Total complete:   28/  87  32%  skipped:   22, failed:    4
ERROR   - qemu_cortex_r5            samples/cpp_synchronization/sample.cpp.synchronization FAILED: Exited with 2
ERROR   - see: /dev/shm/sanity-out/qemu_cortex_r5/samples/cpp_synchronization/sample.cpp.synchronization/build.log
INFO    - Total complete:   29/  87  33%  skipped:   22, failed:    5
ERROR   - qemu_cortex_r5            samples/philosophers/sample.kernel.philosopher.coop_only FAILED: Exited with 2
ERROR   - see: /dev/shm/sanity-out/qemu_cortex_r5/samples/philosophers/sample.kernel.philosopher.coop_only/build.log
INFO    - Total complete:   31/  87  35%  skipped:   22, failed:    6
ERROR   - qemu_cortex_r5            samples/philosophers/sample.kernel.philosopher.fifos FAILED: Exited with 2
ERROR   - see: /dev/shm/sanity-out/qemu_cortex_r5/samples/philosophers/sample.kernel.philosopher.fifos/build.log
INFO    - Total complete:   32/  87  36%  skipped:   22, failed:    7
ERROR   - qemu_cortex_r5            samples/philosophers/sample.kernel.philosopher.stacks FAILED: Exited with 2
ERROR   - see: /dev/shm/sanity-out/qemu_cortex_r5/samples/philosophers/sample.kernel.philosopher.stacks/build.log
INFO    - Total complete:   33/  87  37%  skipped:   22, failed:    8
ERROR   - qemu_cortex_r5            samples/subsys/logging/logger/sample.logger        FAILED: Exited with 2
ERROR   - see: /dev/shm/sanity-out/qemu_cortex_r5/samples/subsys/logging/logger/sample.logger/build.log
INFO    - Total complete:   34/  87  39%  skipped:   22, failed:    9
ERROR   - qemu_cortex_r5            samples/philosophers/sample.kernel.philosopher.preempt_only FAILED: Exited with 2
ERROR   - see: /dev/shm/sanity-out/qemu_cortex_r5/samples/philosophers/sample.kernel.philosopher.preempt_only/build.log
INFO    - Total complete:   35/  87  40%  skipped:   22, failed:   10
ERROR   - qemu_cortex_r5            tests/subsys/logging/log_output/logging.log_output FAILED: Exited with 2
ERROR   - see: /dev/shm/sanity-out/qemu_cortex_r5/tests/subsys/logging/log_output/logging.log_output/build.log
INFO    - Total complete:   36/  87  41%  skipped:   22, failed:   11
ERROR   - qemu_cortex_r5            samples/philosophers/sample.kernel.philosopher.tracing FAILED: Exited with 2
ERROR   - see: /dev/shm/sanity-out/qemu_cortex_r5/samples/philosophers/sample.kernel.philosopher.tracing/build.log
INFO    - Total complete:   37/  87  42%  skipped:   22, failed:   12
ERROR   - qemu_cortex_r5            tests/subsys/debug/tracing/tracing.backends.openocd FAILED: Exited with 2
ERROR   - see: /dev/shm/sanity-out/qemu_cortex_r5/tests/subsys/debug/tracing/tracing.backends.openocd/build.log
INFO    - Total complete:   38/  87  43%  skipped:   22, failed:   13
ERROR   - qemu_cortex_r5            samples/philosophers/sample.kernel.philosopher.lifos FAILED: Exited with 2
ERROR   - see: /dev/shm/sanity-out/qemu_cortex_r5/samples/philosophers/sample.kernel.philosopher.lifos/build.log
INFO    - Total complete:   39/  87  44%  skipped:   22, failed:   14
ERROR   - qemu_cortex_r5            samples/drivers/crypto/sample.drivers.crypto.mbedtls.micro FAILED: Exited with 2
ERROR   - see: /dev/shm/sanity-out/qemu_cortex_r5/samples/drivers/crypto/sample.drivers.crypto.mbedtls.micro/build.log
INFO    - Total complete:   40/  87  45%  skipped:   22, failed:   15
ERROR   - qemu_cortex_r5            samples/application_development/external_lib/sample.app_dev.external_lib FAILED: Exited with 2
ERROR   - see: /dev/shm/sanity-out/qemu_cortex_r5/samples/application_development/external_lib/sample.app_dev.external_lib/build.log
INFO    - Total complete:   41/  87  47%  skipped:   22, failed:   16
ERROR   - qemu_cortex_r5            samples/hello_world/sample.basic.helloworld        FAILED: Exited with 2
ERROR   - see: /dev/shm/sanity-out/qemu_cortex_r5/samples/hello_world/sample.basic.helloworld/build.log
INFO    - Total complete:   42/  87  48%  skipped:   22, failed:   17
ERROR   - qemu_cortex_r5            tests/lib/ringbuffer/libraries.data_structures     FAILED: Exited with 2
ERROR   - see: /dev/shm/sanity-out/qemu_cortex_r5/tests/lib/ringbuffer/libraries.data_structures/build.log
INFO    - Total complete:   43/  87  49%  skipped:   22, failed:   18
ERROR   - qemu_cortex_r5            tests/subsys/logging/log_list/logging.log_list     FAILED: Exited with 2
ERROR   - see: /dev/shm/sanity-out/qemu_cortex_r5/tests/subsys/logging/log_list/logging.log_list/build.log
INFO    - Total complete:   44/  87  50%  skipped:   22, failed:   19
ERROR   - qemu_cortex_r5            tests/subsys/logging/log_msg/logging.log_msg       FAILED: Exited with 2
ERROR   - see: /dev/shm/sanity-out/qemu_cortex_r5/tests/subsys/logging/log_msg/logging.log_msg/build.log
INFO    - Total complete:   45/  87  51%  skipped:   22, failed:   20
ERROR   - qemu_cortex_r5            tests/subsys/logging/log_immediate/logging.log_immediate FAILED: Exited with 2
ERROR   - see: /dev/shm/sanity-out/qemu_cortex_r5/tests/subsys/logging/log_immediate/logging.log_immediate/build.log
INFO    - Total complete:   46/  87  52%  skipped:   22, failed:   21
[... 59 lines cut ...]
INFO    - Total complete:   69/  87  79%  skipped:   22, failed:   41
ERROR   - qemu_cortex_r5            tests/lib/sprintf/libraries.libc.sprintf           FAILED: Exited with 2
ERROR   - see: /dev/shm/sanity-out/qemu_cortex_r5/tests/lib/sprintf/libraries.libc.sprintf/build.log
INFO    - Total complete:   70/  87  80%  skipped:   22, failed:   42
ERROR   - qemu_cortex_r5            tests/subsys/logging/log_core/logging.log_core     FAILED: Exited with 2
ERROR   - see: /dev/shm/sanity-out/qemu_cortex_r5/tests/subsys/logging/log_core/logging.log_core/build.log
INFO    - Total complete:   71/  87  81%  skipped:   22, failed:   43
ERROR   - qemu_cortex_r5            tests/lib/json/libraries.encoding.json             FAILED: Exited with 2
ERROR   - see: /dev/shm/sanity-out/qemu_cortex_r5/tests/lib/json/libraries.encoding.json/build.log
INFO    - Total complete:   74/  87  85%  skipped:   22, failed:   44
ERROR   - qemu_cortex_r5            tests/subsys/shell/shell_history/shell.history     FAILED: Exited with 2
ERROR   - see: /dev/shm/sanity-out/qemu_cortex_r5/tests/subsys/shell/shell_history/shell.history/build.log
INFO    - Total complete:   77/  87  88%  skipped:   22, failed:   45
ERROR   - qemu_cortex_r5            tests/shell/shell.core                             FAILED: Exited with 2
ERROR   - see: /dev/shm/sanity-out/qemu_cortex_r5/tests/shell/shell.core/build.log
INFO    - Total complete:   79/  87  90%  skipped:   22, failed:   46
ERROR   - qemu_cortex_r5            tests/posix/common/portability.posix.common.newlib FAILED: Exited with 2
ERROR   - see: /dev/shm/sanity-out/qemu_cortex_r5/tests/posix/common/portability.posix.common.newlib/build.log
INFO    - Total complete:   81/  87  93%  skipped:   22, failed:   47
ERROR   - qemu_cortex_r5            tests/posix/fs/portability.posix.fs.newlib         FAILED: Exited with 2
ERROR   - see: /dev/shm/sanity-out/qemu_cortex_r5/tests/posix/fs/portability.posix.fs.newlib/build.log
INFO    - Total complete:   87/  87  100%  skipped:   22, failed:   48
INFO    - 17 of 65 tests passed (26.15%), 48 failed, 22 skipped with 0 warnings in 44.21 seconds
INFO    - In total 745 test cases were executed on 1 out of total 218 platforms (0.46%)
```

jhedberg, 2020-01-21:
Will be solved with the new SDK

jhedberg, 2020-01-21:
@galak assigned this to you since it'll be resolved with the new SDK

jhedberg, 2020-02-04:
@galak has this been resolved (i.e. should it be closed) now that we're using the new SDK?

galak, 2020-02-04:
appears to be fixed with SDK 0.11.x release.

closed 2020-02-04 by galak