zephyrproject-rtos/zephyr #22037: qemu_cortex_r5 excludes too many tests

andrewboie, 2020-01-20:
```
$ cat boards/arm/qemu_cortex_r5/qemu_cortex_r5.yaml 
...
  ignore_tags:
    - benchmark
    - cmsis_rtos
    - interrupt
    - kernel
    - memory_protection
    - userspace
```

This target should not be excluding so many categories.
For memory_protection and userspace, these tests already have filters which don't run them if the feature isn't supported (CONFIG_ARCH_HAS_USERSPACE)

interrupt should be enabled or a bug filed on it

kernel should **definitely** be enabled, these are core kernel tests!

cmsis_rtos should be enabled or a bug filed on it

the only category that makes sense here is 'benchmark' since QEMU does not provide useful benchmark data

stephanosio, 2020-02-08:
@andrewboie Most of the above, including the `kernel`, will be enabled once all the major Cortex-R arch port bugs are addressed (see #20594).

stephanosio, 2020-02-10:
**Notes on the `ignore_tags` for the `qemu_cortex_r5`**
* `benchmark`
  * The `benchmark` test implementation for ARM relies on the `SysTick` hardware, which is only present on the Cortex-M devices.
  * List of failed tests: https://gist.github.com/stephanosio/72baa3c03ceb430c74af7a25ff2400d0
  * Failure log: https://gist.github.com/stephanosio/b0233eded3e299a29eb3b932207cedfe
* `interrupt`
  * The `interrupt` test implementation for ARM assumes the availability of `NVIC`, which is only present on the Cortex-M devices.
  * A GIC-based implementation should be implemented for Cortex-R and Cortex-A devices (note that not all Cortex-R devices make use of the GIC, in which case the interrupt test should either be filtered out or executed partially).
  * List of failed tests: https://gist.github.com/stephanosio/3b4e575d04759f436ea9dcd0241b6eb3
  * Failure log: https://gist.github.com/f78db00e94c9dd6b1fe67d18b647cf33
* `memory_protection`
  * MPU support is not implemented in the Cortex-R arch port.
  * This test should be enabled after the MPU implementation is available (consider implementing CMSIS-Core(R) MPU support prior to this)
* `userspace`
  * Userspace support is not implemented in the Cortex-R arch port.
  * This test should be enabled after the userspace implementation is available.

closed 2020-02-11 by andrewboie