zephyrproject-rtos/zephyr #22048: Failing LL.TS Data Length Update Tests (split)

mtpr-ot, 2020-01-20:
**Description**
Failing LL.TS EBQ Tests for Data Length Update:
```
LL/CON/MAS/BV-77-C:
Master Data Length Update - Initiating Data Length Update Procedure; LE 2M PHY

LL/CON/SLA/BV-131-C: 
Data Length Update - Retransmission During an Update

LL/CON/SLA/BV-78-C
Slave Data Length Update - Initiating Data Length Update Procedure; LE 1M PHY

LL/CON/SLA/BV-81-C:
Slave Data Length Update - Initiating Data Length Update Procedure; LE 2M PHY
```
**To Reproduce**
Run the listed LL.TS tests with EBQ and nRF52

**Expected behavior**
Tests should pass

**Impact**
Currently no practical impact, but qualification will fail

mtpr-ot, 2020-01-20:
For `LL/CON/SLA/BV-78-C`, I have a solution proposal which makes the test pass. I'm not 100% sure if this is how to do it:

(call to ll_length_max_get and use of MIN macro added)
```
hci.c:
static void le_set_data_len(struct net_buf *buf, struct net_buf **evt)
{
	struct bt_hci_cp_le_set_data_len *cmd = (void *)buf->data;
	struct bt_hci_rp_le_set_data_len *rp;
	u16_t tx_octets;
	u16_t tx_time;
	u16_t handle;
	u8_t status;

	u16_t max_tx_octets;
	u16_t max_tx_time;
	u16_t max_rx_octets;
	u16_t max_rx_time;

	ll_length_max_get(&max_tx_octets, &max_tx_time,
			  &max_rx_octets, &max_rx_time);

	handle    = sys_le16_to_cpu(cmd->handle);
	tx_octets = MIN(max_tx_octets, sys_le16_to_cpu(cmd->tx_octets));
	tx_time   = MIN(max_tx_time, sys_le16_to_cpu(cmd->tx_time));
	status    = ll_length_req_send(handle, tx_octets, tx_time);

	rp = hci_cmd_complete(evt, sizeof(*rp));
	rp->status = status;
	rp->handle = sys_cpu_to_le16(handle);
}
```

aescolar, 2020-01-20:
CC @cvinayak @joerchan

mtpr-ot, 2020-02-07:
Only the test case left failing on our side:
```
LL/CON/SLA/BV-131-C: 
Data Length Update - Retransmission During an Update
```

kruithofa, 2020-02-24:
Note that feedback from Ellisys is that LL/CON/SLA/BV-131-C is an EBQ related issue, and not a problem in the  LL code

PR #23046 (2020-02-25): Bluetooth: controller: split: fixes DLE conformance tests
For review by Vinayak and EBQ verification by Morten
EBQ run with CODED and 2M phy configured in without failures

Fixes #22048 except for LL/CON/SLA/BV-131-C
See https://github.com/EDTTool/EDTT/pull/12 for a fix for the EDTT
closed 2020-02-25 by aescolar