zephyrproject-rtos/zephyr #22049: Bluetooth: IRK handling issue when using multiple local identities

MarekPieta, 2020-01-20:
**Describe the bug**
When using Bluetooth with multiple local identities, the IRK handling issues may occur.

**To Reproduce**
1) Use Zephyr Bluetooth peripheral with multiple local identities (A, B) and Central that uses IRK.
2) Connect and bond with Central on identity A.
3) Connect and bond with Central on identity B.
        You will see the following error log:
        `<err> bt_hci_core: Failed to add IRK to controller`
        The error log is observable also on every reboot.
        Failure is caused by trying to add the same IRK multiple times.
4) Reset one of the local identities (A or B) - the IRK will be removed from controller causing reconnection problem for second identity. After the reboot, the IRK's are reloaded and the device will be able to reconnect properly.

**Impact**
Error logs and possible reconection problems, when connecting given Central with multiple local identities of the same Zephyr device.

**Environment**
 - Commit SHA: b413223a66ed9e857c67ee21d1e8be4768804dd1

**Additional context**
The problem replicates using various Link Layers.
The IRKs are currently not associated with local identities in the Bluetooth Controller.

joerchan, 2020-01-20:
@jhedberg This looks like an advertising identity issue, similar to #21714
Starting an advertiser should have the IRKs that is valid for it's identity when starting advertising.
Problem can be seen when multiple local identities has a bond with the same remote device.

carlescufi, 2021-01-14:
@joerchan please add a quick line in the documentation about this limitation.

carlescufi, 2022-08-19:
Related to #34848

carlescufi, 2022-08-24:
A new Kconfig option allowing the user to choose between failing the bond procedure or overwriting the bond of the other identity will be introduced, as well as accompanying documentation.

carlescufi, 2022-09-08:
> A new Kconfig option allowing the user to choose between failing the bond procedure or overwriting the bond of the other identity will be introduced, as well as accompanying documentation.

`CONFIG_BT_SAME_PEER_DIFF_ID_BOND_OVERWRITE`:

If set to `y`, then bonding with the same peer but a different identity (when already bonded) will overwrite the existing bond, effectively "moving" the bond from the old to the new identity. Note: this will not be an actual "overwrite" since bonds are identified by `identity:peer address` and thus the new bond will have a different identifier.
If set to `n`, then trying to bond with an already-bonded device but this time with a different identity will result in the pairing procedure failing.

closed 2022-10-27 by carlescufi