zephyrproject-rtos/zephyr #22051: Discovery of 128bit primary service fails with later versions of gcc.

lanvis, 2020-01-20:
**Describe the bug**
Discovery of 128bit primary service fails with later versions of gcc.       
Looks like macro BT_UUID_16 returns wrong value with later gcc versions.

Using gcc ARM version 7.3.1 (BT_UUID_16(uuid)->val) returns 0x2800 (which is correct)

Using gcc ARM version 9.2.1 (BT_UUID_16(uuid)->val) returns 0x2000 (which is incorrect) and leads to a failed discovery.

**To Reproduce**
Attempt a discovery of a 128bit primary service when configured as a central.

A Ellisys trace is attached showing the Find by value request going out as RFCOMM???
[DiscoveryFailGCCNew012020.zip](https://github.com/zephyrproject-rtos/zephyr/files/4087072/DiscoveryFailGCCNew012020.zip)

**Environment (please complete the following information):**
 - OS:  Linux 18.04LTS
 - Toolchain (Zephyr 2.0.0, ARM gcc 9.2.1)

[DiscoveryFailGCCNew012020.zip](https://github.com/zephyrproject-rtos/zephyr/files/4087081/DiscoveryFailGCCNew012020.zip)

lanvis, 2020-01-21:
In gatt_find_type(), BT_UUID_16(uuid)->val) returns wrong value in later gcc versions.

aescolar, 2020-01-21:
CC @jhedberg @Vudentz

joerchan, 2020-01-21:
Since it is Zephyr 2.0.0. This might be already fixed by:
https://github.com/zephyrproject-rtos/zephyr/issues/21186

@lanvis Please check

lanvis, 2020-01-22:
Yep ... that is the same bug.  Sorry about that.  I tested by merging your updates into 2.0 gatt.c file as changes to the file are many.

Many thanks!

closed 2020-01-22 by jhedberg