zephyrproject-rtos/zephyr #22060: Build fails with gnuarmemb under windows

baw-serafin, 2020-01-21:
**Describe the bug**
There is a Toolchain issue with gnuarmemb 9.2 (current version) under Windows

**To Reproduce**
Steps to reproduce the behaviour:
Just follow [getting started](https://docs.zephyrproject.org/latest/getting_started/index.html) and [3rd party toolchain](https://docs.zephyrproject.org/latest/getting_started/toolchain_3rd_party_x_compilers.html#gnu-arm-embedded)

I installed to `c:/gnuarmemb` but I don't think this should matter

**Expected behavior**
'west -b nucleo_f401re samples/hello_world' should work but I get error (see below)

**Impact**
Showstopper for new developers should be easy fixable for people with knowledge to cross compile on windows

**Screenshots or console output**
```
Compiling the C compiler identification source file "CMakeCCompilerId.c" failed.
Compiler: C:/gnuarmemb/bin/arm-none-eabi-gcc.exe
Build flags:
Id flags:  
 
The output was:
1
c:/gnuarmemb/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: c:/gnuarmemb/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/lib\libc.a(lib_a-exit.o): in function `exit':
exit.c:(.text.exit+0x2c): undefined reference to `_exit'
collect2.exe: error: ld returned 1 exit status
 
 
Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" failed.
Compiler: C:/gnuarmemb/bin/arm-none-eabi-gcc.exe
Build flags:
Id flags:  
 
The output was:
1
c:/gnuarmemb/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: c:/gnuarmemb/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/lib\libc.a(lib_a-exit.o): in function `exit':
exit.c:(.text.exit+0x2c): undefined reference to `_exit'
collect2.exe: error: ld returned 1 exit status
```

**Environment (please complete the following information):**
 - Windows
 - gnuarmemb 9.2
 - v2.0.1

SebastianBoe, 2020-01-21:
Could not reproduce.

My env:

```
Windows 10.
The latest gnuarmemb (9.2)
Latest master of zephyr

```

v2.0.1, is that supposed to be a Zephyr revision? I can't find a git tag by that name.

SebastianBoe, 2020-01-21:
Please post full build logs.

baw-serafin, 2020-01-21:
sorry my bad v2.1.0 (latest release)

SebastianBoe, 2020-01-21:
Also, try copying the cache directory to a different directory in case the cache directory is corrupted.

CMake spits out in the logs where the cache directory is on every run.

SebastianBoe, 2020-01-21:
Could not reproduce on v2.0.1 either.

SebastianBoe, 2020-01-27:
Closing as could-not-reproduce.

Feel free to re-open after further investigation is done.

baw-serafin, 2020-01-30:
Worked on another device so probably was a cmake cache problem sorry for the false alert

pfalcon, 2020-06-09:
I'm facing this issue with this toolchain version on Linux. I actually see that I downloaded tarball some time last year, but didn't upgrade our CI, quite possible because I saw this or similar issue. We're now forced to upgrade due to https://github.com/zephyrproject-rtos/zephyr/pull/25804#issuecomment-638704629 .

The situation doesn't seem right - arm-none-eabi-gcc, when passed no options, is unable to link final executable (note that source file, CMakeCCompilerId.c, doesn't have explicit invocation of exit(), which then leads to error of missing _exit()). But still, the question why CMake is doing all this, why CMake tries to invoke compiler in random ways, e.g. Zephyr isn't going to invoke compiler without options, it's going to invoke it with very specific options regarding include paths and libraries, and chances are that with them, everything will work.

@SebastianBoe, @tejlmand, can CMake work in "hands off" approach, where it just does what user tells (e.g., use a toolchain at specific path), instead of trying to outsmart a user and find "errors"? What's the option to tell CMake to do that now?

@galak: FYI

pfalcon, 2020-06-09:
So, despite failure on my system, I decided to upgrade our CI anyway, given that original submitter commented "Worked on another device". And weird as it is, the build went thru: https://ci.linaro.org/view/lite-iot-ci/job/zephyr-upstream/5258/ . 

So, I don't know, let me close this again.

closed 2020-06-09 by pfalcon

ahmedwahdan, 2020-08-13:
I'm facing the same issue
```
C:\Users\ahmed.wahdan\zephyrproject\zephyr>west build -b nucleo_f401re samples/hello_world
-- west build: generating a build system
Including boilerplate (Zephyr base): C:/Users/ahmed.wahdan/zephyrproject/zephyr/cmake/app/boilerplate.cmake
-- Application: C:/Users/ahmed.wahdan/zephyrproject/zephyr/samples/hello_world
-- Zephyr version: 2.3.99 (C:/Users/ahmed.wahdan/zephyrproject/zephyr)
-- Found Python3: C:/Python/Python38/python.exe (found suitable version "3.8.3", minimum required is "3.6") found components: Interpreter
-- Found west: C:/Python/Python38/Scripts/west.exe (found suitable version "0.7.3", minimum required is "0.7.1")
-- Board: nucleo_f401re
-- Found toolchain: gnuarmemb (C:/gnuarmemb)
-- Found BOARD.dts: C:/Users/ahmed.wahdan/zephyrproject/zephyr/boards/arm/nucleo_f401re/nucleo_f401re.dts
-- Generated zephyr.dts: C:/Users/ahmed.wahdan/zephyrproject/zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: C:/Users/ahmed.wahdan/zephyrproject/zephyr/build/zephyr/include/generated/devicetree_unfixed.h
Parsing C:/Users/ahmed.wahdan/zephyrproject/zephyr/Kconfig
Loaded configuration 'C:/Users/ahmed.wahdan/zephyrproject/zephyr/boards/arm/nucleo_f401re/nucleo_f401re_defconfig'
Merged configuration 'C:/Users/ahmed.wahdan/zephyrproject/zephyr/samples/hello_world/prj.conf'
Configuration saved to 'C:/Users/ahmed.wahdan/zephyrproject/zephyr/build/zephyr/.config'
Kconfig header saved to 'C:/Users/ahmed.wahdan/zephyrproject/zephyr/build/zephyr/include/generated/autoconf.h'
-- The C compiler identification is GNU 9.3.1
-- The CXX compiler identification is GNU 9.3.1
-- The ASM compiler identification is GNU
-- Found assembler: C:/gnuarmemb/bin/arm-none-eabi-gcc.exe
-- Cache files will be written to: C:/Users/ahmed.wahdan/zephyrproject/zephyr/.cache
CMake Error at ../../cmake/extensions.cmake:1422 (message):
  Assertion failed: The toolchain is unable to build a dummy C file.  See
  CMakeError.log.
Call Stack (most recent call first):
  ../../CMakeLists.txt:42 (assert)

-- Configuring incomplete, errors occurred!
See also "C:/Users/ahmed.wahdan/zephyrproject/zephyr/build/CMakeFiles/CMakeOutput.log".
See also "C:/Users/ahmed.wahdan/zephyrproject/zephyr/build/CMakeFiles/CMakeError.log".
FATAL ERROR: command exited with status 1: 'C:\Program Files\CMake\bin\cmake.EXE' '-DWEST_PYTHON=c:\python\python38\python.exe' '-BC:\Users\ahmed.wahdan\zephyrproject\zephyr\build' '-SC:\Users\ahmed.wahdan\zephyrproject\zephyr\samples\hello_world' -GNinja -DBOARD=nucleo_f401re
```
CMakeError.log

```
Compiling the C compiler identification source file "CMakeCCompilerId.c" failed.
Compiler: C:/gnuarmemb/bin/arm-none-eabi-gcc.exe 
Build flags: 
Id flags:  

The output was:
1
c:/gnuarmemb/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld.exe: c:/gnuarmemb/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/lib\libc.a(lib_a-exit.o): in function `exit':
exit.c:(.text.exit+0x2c): undefined reference to `_exit'
collect2.exe: error: ld returned 1 exit status

Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" failed.
Compiler: C:/gnuarmemb/bin/arm-none-eabi-gcc.exe 
Build flags: 
Id flags:  

The output was:
1
c:/gnuarmemb/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld.exe: c:/gnuarmemb/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/lib\libc.a(lib_a-exit.o): in function `exit':
exit.c:(.text.exit+0x2c): undefined reference to `_exit'
collect2.exe: error: ld returned 1 exit status
```

ahmedwahdan, 2020-08-13:
Reinstalling the tool chain fixed the issue.

cf1331, 2020-11-06:
I'm having the same issue. It also appears to come up in https://devzone.nordicsemi.com/f/nordic-q-a/55475/the-toolchain-is-unable-to-build-a-dummy-c-file/225042 and https://github.com/platformio/platformio-core/issues/3389.

Based on this thread and the linked threads, I've tried:
- Moving the `build` folder
- Reinstalling the toolchain
- Different toolchain versions (8.3.1, 9.2.1, 9.3.1)
- Adding `set(CMAKE_EXE_LINKER_FLAGS_INIT "--specs=nosys.specs")` to `cmake/toolchain/gnuarmemb/generic.cmake` 

Environment:
- Windows
- gnuarmemb 9.2
- v2.4.0

```
C:\zephyrproject\zephyr>west build -p auto -b nucleo_f429zi samples\basic\blinky
-- west build: generating a build system
Including boilerplate (Zephyr base): C:/zephyrproject/zephyr/cmake/app/boilerplate.cmake
-- Application: C:/zephyrproject/zephyr/samples/basic/blinky
-- Zephyr version: 2.4.0 (C:/zephyrproject/zephyr)
-- Found Python3: C:/Python/3.9.0/python.exe (found suitable exact version "3.9.0") found components: Interpreter
-- Found west (found suitable version "0.8.0", minimum required is "0.7.1")
-- Board: nucleo_f429zi
-- Cache files will be written to: C:/zephyrproject/zephyr/.cache
-- Found toolchain: gnuarmemb (C:/Platforms/arm/toolchain/9-2019-q4)
-- Found BOARD.dts: C:/zephyrproject/zephyr/boards/arm/nucleo_f429zi/nucleo_f429zi.dts
-- Generated zephyr.dts: C:/zephyrproject/zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: C:/zephyrproject/zephyr/build/zephyr/include/generated/devicetree_unfixed.h
Parsing C:/zephyrproject/zephyr/Kconfig
Loaded configuration 'C:/zephyrproject/zephyr/build/zephyr/.config'
No change to configuration in 'C:/zephyrproject/zephyr/build/zephyr/.config'
No change to Kconfig header in 'C:/zephyrproject/zephyr/build/zephyr/include/generated/autoconf.h'
-- The C compiler identification is GNU 9.2.1
-- The CXX compiler identification is GNU 9.2.1
-- The ASM compiler identification is GNU
-- Found assembler: C:/Platforms/arm/toolchain/9-2019-q4/bin/arm-none-eabi-gcc.exe
CMake Error at ../../../cmake/extensions.cmake:1557 (message):
  Assertion failed: The toolchain is unable to build a dummy C file.  See
  CMakeError.log.
Call Stack (most recent call first):
  ../../../CMakeLists.txt:42 (assert)

-- Configuring incomplete, errors occurred!
See also "C:/zephyrproject/zephyr/build/CMakeFiles/CMakeOutput.log".
See also "C:/zephyrproject/zephyr/build/CMakeFiles/CMakeError.log".
FATAL ERROR: command exited with status 1: 'C:\Program Files\CMake\bin\cmake.EXE' '-DWEST_PYTHON=c:\python\3.9.0\python.exe' '-BC:\zephyrproject\zephyr\build' '-SC:\zephyrproject\zephyr\samples\basic\blinky' -GNinja -DBOARD=nucleo_f429zi
```

CMakeError.log
```
Compiling the C compiler identification source file "CMakeCCompilerId.c" failed.
Compiler: C:/Platforms/arm/toolchain/9-2019-q4/bin/arm-none-eabi-gcc.exe 
Build flags: 
Id flags:  

The output was:
1
c:/platforms/arm/toolchain/9-2019-q4/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: c:/platforms/arm/toolchain/9-2019-q4/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/lib\libc.a(lib_a-exit.o): in function `exit':
exit.c:(.text.exit+0x2c): undefined reference to `_exit'
collect2.exe: error: ld returned 1 exit status

Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" failed.
Compiler: C:/Platforms/arm/toolchain/9-2019-q4/bin/arm-none-eabi-gcc.exe 
Build flags: 
Id flags:  

The output was:
1
c:/platforms/arm/toolchain/9-2019-q4/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: c:/platforms/arm/toolchain/9-2019-q4/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/lib\libc.a(lib_a-exit.o): in function `exit':
exit.c:(.text.exit+0x2c): undefined reference to `_exit'
collect2.exe: error: ld returned 1 exit status

```

tejlmand, 2020-11-09:
@cf1331 This looks suspicious to me:
> c:/platforms/arm/toolchain/9-2019-q4/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/lib\libc.a

Mixing of path separators is usually a bad sign.
Will try to reproduce.

dzarda, 2020-11-09:
Seeing the same issue when running from within PlatformIO.

Fixed by `int _exit(){}`

leonahi, 2020-11-20:
I am facing the same issue as indicated by @cf1331 but with ubuntu 14.04 environment. Reinstalling the tool chain didn't solved the problem.

nextcoffee, 2020-11-24:
I've encountered this issue when I use the latest version of CMake (v3.19.0), once I downgrade CMake to v3.16.2, the build process is finished without any errors. BTW, I work on windows 10.

mmccoo, 2020-11-24:
After downgrading CMake, build finishes for me as well.

But...

While I get an elf file, the CMakeError.log file still complains about _exit. I have not yet tried to flash to a board.

I'm on arc linux

rhssk, 2020-11-24:
The first point seems relevant:
https://cmake.org/cmake/help/latest/release/3.19.html#id1

So either downgrade to 3.18 or upgrade to 3.19.1.

vrahane, 2020-11-24:
I had a similar issue, what worked for me was installing CMake release version 3.17.3 from https://cmake.org/files/ and then created a symlink to the executable. What did not work was `cmake/3.19.0_1` installed via homebrew.

zijinhulu, 2020-11-25:
I'm having the same issue. in window10.when I delete the contents of the "zephyr\.cache\ToolchainCapabilityDatabase" folder, the problem is solved

tejlmand, 2020-11-25:
This is something that should be improved, but without sacrifying the benefits of the toolchain capabilities database.
Created: https://github.com/zephyrproject-rtos/zephyr/issues/30227 for followup.

galak, 2020-11-25:
cmake 3.19 and 3.19.1 currently don't work with Zephyr.  This appears to be a regression with upstream CMake and I've opened the following issue to track things:

https://github.com/zephyrproject-rtos/zephyr/issues/30232

utsavm9, 2020-11-25:
Like last comment, this worked for me on Windows 10 (Powershell):
```
choco install cmake --installargs 'ADD_CMAKE_TO_PATH=System' --version=3.17.2 --allow-downgrade
rm -r build
rm -r .cache
```

vinravind, 2021-03-26:
Removing the zephyr cache before west build fixed the probem