zephyrproject-rtos/zephyr #22061: Ethernet switch support

parthitce, 2020-01-21:
**Is your feature request related to a problem? Please describe.**
I am currently involved in adding support for [embOS/IP Switch Board](https://www.segger.com/evaluate-our-software/segger/embosip-switch-board/) based on MK66FN2M0VMD18 to Zephyr. The board also comes with Micrel/Microchip KSZ8794CNX 3 port switch. To add support for the switch or to make Zephyr for Switch devices, there is no existing framework/sub-sys support in Ethernet driver.

**Describe the solution you'd like**
I have used similar switch devices under Linux as Distributed Switch Architecture (DSA) which helps in configuring and describing the ports (from device tree). Is it possible to implement or port DSA for Zephyr?

**Describe alternatives you've considered**
TCP/IP stacks like Cyclone TCP already have support for switches along with Ethernet, PHY controllers. Does it makes sense to port it instead of writing it based in DSA?

**Additional context**
- I am not sure if there is already open PR/Issue for this request or support is already available and I am missing it.
- If no WIP, I would like to start this based on inputs for this issue.

carlescufi, 2020-03-31:
@jukkar and @rlubos could on you comment on the questions/requirements in the description?

jukkar, 2020-04-01:
Link to DSA in Linux https://www.kernel.org/doc/html/latest/networking/dsa/dsa.html

lmajewski, 2020-04-16:
I've prepared the RFC PR which is a preparatory work for adding KSZ8794 DSA switch.
Please refer to https://github.com/zephyrproject-rtos/zephyr/pull/24323 . Despite, I'm going to divide this PR to smaller pieces, comments are always more than welcome.

jukkar, 2021-03-06:
DSA support is already merged, so closing this one.

closed 2021-03-06 by jukkar