zephyrproject-rtos/zephyr #22062: Adafruit Feather M0 does not flash correctly - incorrect flash code offset and bossa version incompatibility

KubaFYI, 2020-01-21:
**Describe the bug**
Adafruit Feather M0 does not flash correctly.

- `west flash` complains about bossa not liking the offset pameter `-o`
- running the bossa command manually without the offending parameter leads to it reporting a successful flash and verify but the board still does not run the code

**To Reproduce**
Steps to reproduce the behavior:
1. `west build -b adafruit_feather_m0_basic_proto samples/basic/blinky --pristine`
2. `west flash`

**Expected behavior**
The board flashes and runs the blinky code.

**Impact**
Showstopper - can't develop with west with default config.

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK 0.10.3 and 0.11.0-alpha-8
 - Zephyr Version: 2.1.99

**Additional context**

I've managed to partially resolve the two parts of the problem with those tweaks:

1. Added `CONFIG_USE_DT_CODE_PARTITION=y` to `adafruit_deather_m0_masic_proto_defconfig`.

The adafruit boards ship with a protected bootloader expecting the code to start at 0x2000. Setting `FLASH_LOAD_OFFSET` to 0x2000 also seems to do the job but after perusing threads of various issues it seems like this is the cleaner way of forcing the device tree-defined code partition to actually be linked where it belongs.

2. Using bossa version 1.9.1 instead.

It seems like previous version of bossa don't implement the necessary `--offset` parameter. The SDK actually used to use 1.9.1 but that was reverted to 1.7 by zephyrproject-rtos/sdk-ng#68 because an upgrade to 1.9.1 stopped Arduino Due from flashing correctly. My understanding is that there was a change in how bossa approached offsets in that starting with 1.9 it needs them explicitly specified while earlier versions assumed certain values. I feel like an appropriate way to go would be to bump it up to 1.9.1 once again and fix up the Arduino Due config instead so that it handles code offsetting correctly?

3. Bossa runner needs to actually be passed the correct offset value

It seems like currently even with the two above `west flash` still doesn't flash correctly. This is because the correct offset is at no point passed down to the bossa runner and so it uses it's default argparse value of 0. I'm not sure where would be an appropriate place to get that value from for west flash. Ideas?

henrikbrixandersen, 2020-01-22:
Thank you for reporting this. It is not specific to the Adafruit Feather M0 Basic Proto board.

The issue arises since the version of `bossac` shipped in the Zephyr SDK no longer supports `-o` for specifying an offset, which is required by the Zephyr bossac west runner in `scripts/west_commands/runners/bossac.py`.

The version of bossac was reverted by @galak in zephyrproject-rtos/sdk-ng/pull/68 but the west runner was not updated. I believe we need to get back to a more recent version of bossa and fix any incompatibilities in the command line.

KubaFYI, 2020-01-22:
I know it's not specific to Adafruit Feather but that's how I came across those several problems so I thought I'd bundle them together. I'm aware that the version of `bossac` is a part of the problem (as mentioned in the original issue text) but just clearing out the mismatch between `bossac` and bossac west runner won't fix the issue. `USE_DT_CODE_PARTITION` has to be added to Feather's defconfig. Should I go ahead and do it?

henrikbrixandersen, 2020-01-28:
@KubaFYI I take it this can be closed now?

henrikbrixandersen, 2020-02-06:
Assuming this is fixed. Please reopen if not so.

closed 2020-02-06 by henrikbrixandersen