zephyrproject-rtos/zephyr #22063: fs/NVS: NVS is not compatible with flash memory's which have 0x00 as erased

nvlsianpu, 2020-01-21:
**This is informative issue**, not a bu really
NVS design assumes 0xff as memory erased state. So it will doesn't work on devices which has 0x00 as erased byte state (e.g. some stm32 embedded and native_posix with the flash simulator)

**To Reproduce**
Try NVS sample using native_posix target

carlescufi, 2020-01-21:
FYI @Laczen

Laczen, 2020-01-21:
In general it would be good (and trivial) to add support in nvs for devices that use 0x00 as erase value. This however would not solve the problem with flash simulation on native posix. In the flash simulator on native posix the flash area is filled with 0x00 at start but is set to 0xff after erase. For the flash simulator on native posix it would be better to erase (set to 0xff) on creation of flash.bin.

de-nordic, 2020-06-26:
The solution https://github.com/zephyrproject-rtos/zephyr/pull/25947 does not seem to work correctly.

Laczen, 2020-06-26:
@de-nordic , might this be caused by line 417 in flash_simulator.c that initializes flash to 0xff ?

de-nordic, 2020-06-26:
> 
> 
> @de-nordic , might this be caused by line 417 in flash_simulator.c that initializes flash to 0xff ?

Thanks @Laczen! That may be the case, I will check this.

de-nordic, 2020-06-26:
> 
> > @de-nordic , might this be caused by line 417 in flash_simulator.c that initializes flash to 0xff ?
> 
There was a problem, but still this is not the root cause.

Laczen, 2020-06-26:
> > > @de-nordic , might this be caused by line 417 in flash_simulator.c that initializes flash to 0xff ?
> 
> There was a problem, but still this is not the root cause.

Do you have an idea where it fails?

de-nordic, 2020-06-26:
> 
> 
> > > > @de-nordic , might this be caused by line 417 in flash_simulator.c that initializes flash to 0xff ?
> > 
> > 
> > There was a problem, but still this is not the root cause.
> 
> Do you have an idea where it fails?

I narrowed it to `nvs_read_hist`, where it returns `-ENOENT`, due all conditions evaluating to 1.

I still do not understand operation of NVS good enough to write more on the subject.

Laczen, 2020-06-26:
@de-nordic, I am a bit puzzled, did you run the test before the merge ? -> I was looking at what would be different from your local test and the automated test.

de-nordic, 2020-06-26:
> 
> 
> @de-nordic, I am a bit puzzled, did you run the test before the merge ? -> I was looking at what would be different from your local test and the automated test.

Yes I did. I run both of the nvs tests each time I have updated something. I do not know why it did pass prior.

Laczen, 2020-06-26:
> > @de-nordic, I am a bit puzzled, did you run the test before the merge ? -> I was looking at what would be different from your local test and the automated test.
> 
> Yes I did. I run both of the nvs tests each time I have updated something. I do not know why it did pass prior.

OK, maybe it is in the 0x00 overlay file (< 0x00 > instead of <0x00> ?)

de-nordic, 2020-06-26:
I have printed out the values that are used for initialization of simulated flash and they are 0.
```
Booting from ROM..Optimal CONFIG_X86_MMU_PAGE_POOL_PAGES 6
Intitlizing to 0
*** Booting Zephyr OS build zephyr-v2.3.0-607-g16763ff9b6db  ***
Running test suite test_nvs
===================================================================
START - test_nvs_init
I: 5 Sectors of 1024 bytes
I: alloc wra: 0, 3f0
I: data wra: 0, 0
 PASS - test_nvs_init
===================================================================
START - test_nvs_write
I: 5 Sectors of 1024 bytes
I: alloc wra: 0, 3f0
I: data wra: 0, 0
wlk 3f0 3f0
wlk.id 0 1
wlk_ate.len 0
cnt 0 0
1
1
wlk 3e8 3e8
wlk.id 0 1
wlk_ate.len 0
cnt 0 0
1
1

    Assertion failed at WEST_TOPDIR/zephyr/tests/subsys/fs/nvs/src/main.c:100: execute_long_pattern_write: (len == sizeof(rd_buf) is false)
nvs_read unexpected failure: -2
 FAIL - test_nvs_write
```

de-nordic, 2020-06-26:
`Intitlizing to 0` comes from the flash driver.

de-nordic, 2020-06-26:
Same run but on the non-0x00 test:
```
ESC[0mSeaBIOS (version rel-1.12.1-0-ga5cab58-dirty-20200202_184651-8999447e980d-zephyr
)
Booting from ROM..Optimal CONFIG_X86_MMU_PAGE_POOL_PAGES 6
Intitlizing to ff
*** Booting Zephyr OS build zephyr-v2.3.0-607-g16763ff9b6db  ***
Running test suite test_nvs
===================================================================
START - test_nvs_init
I: 5 Sectors of 1024 bytes
I: alloc wra: 0, 3f0
I: data wra: 0, 0
 PASS - test_nvs_init
===================================================================
START - test_nvs_write
I: 5 Sectors of 1024 bytes
I: alloc wra: 0, 3f0
I: data wra: 0, 0
wlk 3f0 3f0
wlk.id ffff 1
wlk_ate.len ffff
cnt 0 0
1
0
wlk 3e8 3e8
wlk.id 1 1
wlk_ate.len 200
cnt 1 0
0
0
 PASS - test_nvs_write
```

Laczen, 2020-06-26:
@de-nordic, it seems that even the first write fails, in the flash_simulator there is a suspicious line at 261-262:

```
/* only pull bits to zero */
*(FLASH(offset + i)) &= *((uint8_t *)data + i);
```
shouldn't this be changed ?

de-nordic, 2020-06-26:
@Laczen Yes, you have identified the problem correctly. This is the line. NVS requires no changes. I will fix this and extend testcases for flash simulator.

de-nordic, 2020-07-24:
This issue has been solved, closing.

PR #25947 (2020-06-22): drivers/flash: Add flash_get_parameters call to flash API
Provided call returns const pointer to flash_parameters structure that can be used for passing memory parameters from drivers.

Resolves #22063
closed 2020-07-24 by de-nordic