zephyrproject-rtos/zephyr #22066: tests/kernel/mem_pool/mem_pool_threadsafe fails reliably on m2gl025_miv

andrewboie, 2020-01-21:
**Describe the bug**
```
$ cat /dev/shm/sanity-out/m2gl025_miv/tests/kernel/mem_pool/mem_pool_threadsafe/kernel.memory_pool.threadsafe/handler.log
09:24:06.8875 [INFO] Loaded monitor commands from: /opt/renode/./scripts/monitor.py
09:24:06.9232 [INFO] Including script: /home/apboie/projects/zephyr/zephyr/boards/riscv/m2gl025_miv/support/m2gl025_miv.resc
09:24:06.9320 [INFO] System bus created.
09:24:07.3738 [DEBUG] Segment size automatically calculated to value 64KiB
09:24:07.3743 [DEBUG] Segment size automatically calculated to value 64KiB
09:24:07.3744 [DEBUG] Segment size automatically calculated to value 64KiB
09:24:07.3745 [DEBUG] Segment size automatically calculated to value 4MiB
09:24:07.4771 [WARNING] Writing to an output GPIO pin #0
09:24:07.4772 [WARNING] Writing to an output GPIO pin #1
09:24:07.4772 [WARNING] Writing to an output GPIO pin #2
09:24:07.6968 [DEBUG] sysbus: Loading ELF /dev/shm/sanity-out/m2gl025_miv/tests/kernel/mem_pool/mem_pool_threadsafe/kernel.memory_pool.threadsafe/zephyr/zephyr.elf.
09:24:07.7016 [INFO] sysbus: Loading segment of 25252 bytes length at 0x80000000.
09:24:07.7078 [DEBUG] sysbus: Segment loaded.
09:24:07.7079 [INFO] sysbus: Loading segment of 208 bytes length at 0x800062A4.
09:24:07.7079 [DEBUG] sysbus: Segment loaded.
09:24:07.7080 [INFO] sysbus: Loading segment of 8000 bytes length at 0x800400D0.
09:24:07.7080 [DEBUG] sysbus: Segment loaded.
09:24:07.7335 [INFO] cpu: Setting PC value to 0x80000000.
09:24:07.7778 [INFO] Mi-V: Machine started.
09:24:07.8148 [WARNING] plic: Unhandled write to offset 0x200000, value 0x0.
09:24:07.8194 [WARNING] uart: Unhandled write to offset 0xC. Unhandled bits: [0] when writing value 0x1. Tags: BIT8 (0x1).
09:24:07.8292 [DEBUG] uart: [+0.32s host +0s virt 0s virt from start] *** Booting Zephyr OS build zephyr-v2.1.0-1236-gcf64c6bccaba  ***
09:24:07.8359 [DEBUG] uart: [+6.79ms host +0.1ms virt 0.1ms virt from start]   Running test suite mpool_threadsafe
09:24:07.8366 [DEBUG] uart: [+0.63ms host +0s virt 0.1ms virt from start]   ===================================================================
09:24:07.8370 [DEBUG] uart: [+0.42ms host +0s virt 0.1ms virt from start]   starting test - test_mpool_threadsafe
09:24:07.8514 [DEBUG] uart: [+14.45ms host +1.2ms virt 1.3ms virt from start]   @ ZEPHYR_BASE/lib/os/assert.c:45
09:24:07.8521 [DEBUG] uart: [+0.67ms host +0s virt 1.3ms virt from start]   E: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
09:24:07.8532 [DEBUG] uart: [+1.08ms host +0s virt 1.3ms virt from start]   E: Current thread: 0x80040300 (unknown)
09:24:07.8537 [DEBUG] uart: [+0.5ms host +0.1ms virt 1.4ms virt from start]   E: Halting system
```

**To Reproduce**
1. Install mono and renode emulator:
https://github.com/renode/renode/releases/tag/v1.8.2
https://www.mono-project.com/download/stable/#download-lin
2. Build and run test/kernel/mem_pool/mem_pool_threadsafe for the m2gl025_miv platform

**Expected behavior**
Passing test.

**Impact**
Not sure what the root cause of this is.

nashif, 2020-01-21:
@andrewboie  I am not able to reproduce this locally..

```
i9:zephyr(master): west build -b m2gl025_miv tests/kernel/mem_pool/mem_pool_threadsafe -t run
-- west build: build configuration:
       source directory: /home/nashif/Work/zephyrproject/zephyr/tests/kernel/mem_pool/mem_pool_threadsafe
       build directory: /home/nashif/Work/zephyrproject/zephyr/build
       BOARD: m2gl025_miv (origin: CMakeCache.txt)
-- west build: running target run
[0/1] cd /home/nashif/Work/zephyrproject/zephyr/build && /usr/bin/renod...hyrproject/zephyr/boards/riscv/m2gl025_miv/support/m2gl025_miv.resc; s'
14:00:46.0120 [INFO] Including script: /home/nashif/Work/zephyrproject/zephyr/boards/riscv/m2gl025_miv/support/m2gl025_miv.resc
14:00:46.0200 [INFO] System bus created.
14:00:46.5077 [DEBUG] Segment size automatically calculated to value 64KiB (3)
14:00:46.5078 [DEBUG] Segment size automatically calculated to value 4MiB
14:00:46.6145 [WARNING] Writing to an output GPIO pin #0
14:00:46.6146 [WARNING] Writing to an output GPIO pin #1
14:00:46.6147 [WARNING] Writing to an output GPIO pin #2
14:00:46.8541 [DEBUG] sysbus: Loading ELF /home/nashif/Work/zephyrproject/zephyr/build/zephyr/zephyr.elf.
14:00:46.8592 [INFO] sysbus: Loading segment of 25396 bytes length at 0x80000000.
14:00:46.8659 [DEBUG] sysbus: Segment loaded.
14:00:46.8660 [INFO] sysbus: Loading segment of 208 bytes length at 0x80006334.
14:00:46.8661 [DEBUG] sysbus: Segment loaded.
14:00:46.8661 [INFO] sysbus: Loading segment of 8000 bytes length at 0x800400D0.
14:00:46.8662 [DEBUG] sysbus: Segment loaded.
14:00:46.8880 [INFO] cpu: Setting PC value to 0x80000000.
14:00:46.9279 [INFO] Mi-V: Machine started.
14:00:46.9854 [WARNING] plic: Unhandled write to offset 0x200000, value 0x0.
14:00:46.9905 [WARNING] uart: Unhandled write to offset 0xC. Unhandled bits: [0] when writing value 0x1. Tags: BIT8 (0x1).
14:00:46.9969 [DEBUG] uart: [+0.35s host +0s virt 0s virt from start] *** Booting Zephyr OS build zephyr-v2.1.0-1236-gcf64c6bccaba  ***
14:00:47.0075 [DEBUG] uart: [+10.61ms host +0.1ms virt 0.1ms virt from start]   Running test suite mpool_threadsafe
14:00:47.0091 [DEBUG] uart: [+1.6ms host +0s virt 0.1ms virt from start]   ===================================================================
14:00:47.0101 [DEBUG] uart: [+1.02ms host +0s virt 0.1ms virt from start]   starting test - test_mpool_threadsafe
14:00:47.0721 [DEBUG] uart: [+61.9ms host +19.7ms virt 19.8ms virt from start]   PASS - test_mpool_threadsafe
14:00:47.0748 [DEBUG] uart: [+2.86ms host +0.1ms virt 19.9ms virt from start]   ===================================================================
14:00:47.0759 [DEBUG] uart: [+1.09ms host +0s virt 19.9ms virt from start]   Test suite mpool_threadsafe succeeded
14:00:47.0778 [DEBUG] uart: [+1.8ms host +0s virt 19.9ms virt from start]   ===================================================================
14:00:47.0793 [DEBUG] uart: [+1.6ms host +0.1ms virt 20ms virt from start]   PROJECT EXECUTION SUCCESSFUL
```

nashif, 2020-02-06:
can't reproduce

closed 2020-02-06 by nashif