zephyrproject-rtos/zephyr #22079: add reception channel information to advertise_report

sbiwa-where, 2020-01-22:
Please add reception channel information to advertise_report.
For positioning using advertise_report, we usually use RSSI values. However, the RSSI value varies depending on the receiving channel, resulting in poor accuracy.
We need the channel information received to improve accuracy. (This function exists in NORDIC nRF5 SDK)
Please consider it.

aescolar, 2020-01-22:
CC @cvinayak @jhedberg

jhedberg, 2020-01-22:
@carlescufi @joerchan @trond-snekvik FYI

Zephyr uses standard HCI between the host and controller, and unfortunately neither the legacy Advertising Report HCI Event nor the Extended Advertising Report HCI event contain channel information. This means that for such a feature to be introduced we would need a vendor event to convey the information.

joerchan, 2020-01-22:
CC: @rugeGerritsen 

This is something that is not supported by HCI, and the nRF5 SDK has an easier time going around HCI than we do in Zephyr.
It is also other instances of similar changes that would complement each other that should be considered as part of a complete feature. So this is not a trivial addition.

We can take it under consideration.

kruithofa, 2022-08-25:
this is a proprietry feature,  we can reopen if a new request for this comes

closed 2022-08-25 by kruithofa