zephyrproject-rtos/zephyr #22083: stm32: spi: Infinite loop of RXNE bit check

KwonTae-young, 2020-01-22:
**Describe the bug**
I have often experienced board hangs when using LoRa on STM32L151CBT.
If gdb checks while the board is stopped, `RXNE` continues to be checked.  
https://github.com/zephyrproject-rtos/zephyr/blob/master/drivers/spi/spi_ll_stm32.c#L114-L116
``` c
while (!ll_func_rx_is_not_empty(spi)) {
        /* NOP */
}
```

**To Reproduce**
```
west init zephyrproject_spi_rxne -m https://github.com/KwonTae-young/zephyr/ --mr stm32_spi_rxne
cd zephyrproject_spi_rxne
west update
cd zephyr
source zephyr.sh
west build -b 96b_wistrio samples/drivers/lora/send
west flash -r stm32flash --device /dev/ttyUSB0
```

**Expected behavior**
The board should not stop.

**Impact**
When using LoRa, the board hangs at random times.

**Screenshots or console output**
1. console
```
*** Booting Zephyr OS build zephyr-v1.13.0-13521-ga22c423b115e  ***
[00:00:00.222,000] <inf> sx1276: SX1276 Version:12 found
[00:00:00.232,000] <inf> lora_send: Data sent!
[00:00:00.243,000] <inf> lora_send: Data sent!
[00:00:00.254,000] <inf> lora_send: Data sent!
...
...
[00:00:03.328,000] <inf> lora_send: Data sent!                                  
[00:00:03.339,000] <inf> lora_send: Data sent!                                  
[00:00:03.350,000] <inf> lora_send: Data sent!                                  
[00:00:03.361,000] 
```

2. gdb
```
Reading symbols from /work/zephyrproject_spi_rxne/zephyr/samples/drivers/lora/send/build/zephyr/zephyr.elf...done.
(gdb) bt
#0  LL_SPI_IsActiveFlag_RXNE (SPIx=0x40013000)
    at /work/zephyrproject_spi_rxne/modules/hal/stm32/stm32cube/stm32l1xx/drivers/include/stm32l1xx_ll_spi.h:739
#1  ll_func_rx_is_not_empty (spi=0x40013000)
    at /work/zephyrproject_spi_rxne/zephyr/drivers/spi/spi_ll_stm32.h:40
#2  spi_stm32_shift_m (data=<optimized out>, spi=<optimized out>)
    at /work/zephyrproject_spi_rxne/zephyr/drivers/spi/spi_ll_stm32.c:114
#3  spi_stm32_shift_frames (data=0x20001e20 <spi_stm32_dev_data_1>, 
    spi=<optimized out>)
    at /work/zephyrproject_spi_rxne/zephyr/drivers/spi/spi_ll_stm32.c:179
#4  transceive (asynchronous=false, signal=0x0, rx_bufs=<optimized out>, 
    tx_bufs=0x20000cc8 <z_main_stack+864>, config=0x20000268 <dev_data+12>, 
    dev=<optimized out>)
    at /work/zephyrproject_spi_rxne/zephyr/drivers/spi/spi_ll_stm32.c:434
#5  spi_stm32_transceive (dev=<optimized out>, 
    config=0x20000268 <dev_data+12>, tx_bufs=0x20000cc8 <z_main_stack+864>, 
    rx_bufs=<optimized out>)
    at /work/zephyrproject_spi_rxne/zephyr/drivers/spi/spi_ll_stm32.c:457
#6  0x08006b12 in z_impl_spi_transceive (config=0x20000268 <dev_data+12>, 
    rx_bufs=0x0, tx_bufs=0x20000cc8 <z_main_stack+864>, dev=<optimized out>)
    at /work/zephyrproject_spi_rxne/zephyr/include/drivers/spi.h:263
#7  spi_transceive (config=0x20000268 <dev_data+12>, rx_bufs=0x0, 
    tx_bufs=0x20000cc8 <z_main_stack+864>, dev=<optimized out>)
--Type <RET> for more, q to quit, c to continue without paging--
    at /work/zephyrproject_spi_rxne/zephyr/samples/drivers/lora/send/build/zephyr/include/generated/syscalls/spi.h:33
#8  spi_write (config=0x20000268 <dev_data+12>, 
    tx_bufs=0x20000cc8 <z_main_stack+864>, dev=<optimized out>)
    at /work/zephyrproject_spi_rxne/zephyr/include/drivers/spi.h:303
#9  sx1276_transceive (length=<optimized out>, data=<optimized out>, 
    write=true, reg=<optimized out>)
    at /work/zephyrproject_spi_rxne/zephyr/drivers/lora/sx1276.c:216
#10 sx1276_write (reg_addr=<optimized out>, data=<optimized out>, 
    len=<optimized out>)
    at /work/zephyrproject_spi_rxne/zephyr/drivers/lora/sx1276.c:226
#11 0x08006b26 in SX1276WriteBuffer (addr=<optimized out>, 
    buffer=<optimized out>, size=<optimized out>)
    at /work/zephyrproject_spi_rxne/zephyr/drivers/lora/sx1276.c:233
#12 0x08009920 in SX1276Write (addr=addr@entry=17, data=<optimized out>, 
    data@entry=247 '\367')
    at /work/zephyrproject_spi_rxne/modules/lib/loramac-node/src/radio/sx1276/sx1276.c:1258
#13 0x08006fdc in SX1276SetTx (timeout=<optimized out>)
    at /work/zephyrproject_spi_rxne/modules/lib/loramac-node/src/radio/sx1276/sx1276.c:1085
#14 0x08007152 in SX1276Send (
    buffer=buffer@entry=0x20001ea8 <data> "helloworldGPIOA", 
--Type <RET> for more, q to quit, c to continue without paging--
    size=<optimized out>, size@entry=10 '\n')
    at /work/zephyrproject_spi_rxne/modules/lib/loramac-node/src/radio/sx1276/sx1276.c:850
#15 0x08009854 in sx1276_lora_send (dev=<optimized out>, 
    data=0x20001ea8 <data> "helloworldGPIOA", data_len=<optimized out>)
    at /work/zephyrproject_spi_rxne/zephyr/drivers/lora/sx1276.c:325
#16 0x080041dc in lora_send (data=0x20001ea8 <data> "helloworldGPIOA", 
    data_len=10, dev=0x20001fa8 <__device_sx1276_lora>)
    at /work/zephyrproject_spi_rxne/zephyr/include/drivers/lora.h:115
#17 main ()
    at /work/zephyrproject_spi_rxne/zephyr/samples/drivers/lora/send/src/main.c:48
(gdb) 
(gdb) x /9x 0x40013000
0x40013000:	0x00000324	0x00000000	0x00000000	0x0000008b
0x40013010:	0x00000007	0x00000000	0x00000000	0x00000000
0x40013020:	0x00000000
(gdb) 
```

**Environment**
 - OS: Ubuntu 18.04
 - Toolchain: Zephyr SDK(0.10.3) 
 - Commit SHA or Version used: 9fe1e2a66378ffbfbbd0c08cf27859db80cf0c0d

**Additional context**
The test program modified the delay to 10m for quick symptom reproduction.
The time the board stops is random.
If symptoms do not appear within 1 minute, the symptoms do not seem to be reproduced.
So I test for 1 minute and reset the power.

KwonTae-young, 2020-01-30:
Looking at the gdb data, the `SPE` bit in the `SPI_CR1` register is disabled.
I'm not sure why the `SPE bit` is cleared.
[image]

I modified the source as below.
After testing, the board does not stop.
But I don't know if this is the right way. 
``` diff
diff --git a/drivers/spi/spi_ll_stm32.c b/drivers/spi/spi_ll_stm32.c
index 6d15a4fc39..b81444307e 100644
--- a/drivers/spi/spi_ll_stm32.c
+++ b/drivers/spi/spi_ll_stm32.c
@@ -112,7 +112,10 @@ static void spi_stm32_shift_m(SPI_TypeDef *spi, struct spi_stm32_data *data)
        }
 
        while (!ll_func_rx_is_not_empty(spi)) {
-               /* NOP */
+               if (!LL_SPI_IsEnabled(spi)) {
+                       LOG_INF("SPI Enable");
+                       LL_SPI_Enable(spi);
+               }
        }
 
        if (SPI_WORD_SIZE_GET(data->ctx.config->operation) == 8) {

```
```
[00:00:15.465,000] <inf> lora_send: Data sent!
[00:00:15.476,000] <inf> lora_send: Data sent!
[00:00:15.487,000] <inf> lora_send: Data sent!
[00:00:15.498,000] <inf> lora_send: Data sent!
[00:00:15.509,000] <inf> spi_ll_stm32: SPI Enable
[00:00:15.510,000] <inf> lora_send: Data sent!
[00:00:15.522,000] <inf> lora_send: Data sent!
[00:00:15.533,000] <inf> lora_send: Data sent!
[00:00:15.544,000] <inf> lora_send: Data sent!
```

erwango, 2020-01-30:
@KwonTae-young , this look indeed

For the bus fault, this might be due to another issue. Can you have a try to limit sysfreq to 16MHz? 
cf https://github.com/zephyrproject-rtos/zephyr/pull/22308 (not the final fix of course)

KwonTae-young, 2020-01-31:
@erwango
The following changes were made to test at 16 MHz.
``` diff
diff --git a/boards/arm/96b_wistrio/96b_wistrio_defconfig b/boards/arm/96b_wistrio/96b_wistrio_defconfig
index b12fa7c309..b7dd300586 100644
--- a/boards/arm/96b_wistrio/96b_wistrio_defconfig
+++ b/boards/arm/96b_wistrio/96b_wistrio_defconfig
@@ -19,7 +19,7 @@ CONFIG_CLOCK_CONTROL=y
 CONFIG_CLOCK_STM32_SYSCLK_SRC_PLL=y
 # Use HSI source
 CONFIG_CLOCK_STM32_PLL_SRC_HSI=y
-CONFIG_CLOCK_STM32_PLL_DIVISOR=2
+CONFIG_CLOCK_STM32_PLL_DIVISOR=4
 # produce 32Mhz clock at PLL output
 CONFIG_CLOCK_STM32_PLL_MULTIPLIER=4
 CONFIG_CLOCK_STM32_AHB_PRESCALER=1
diff --git a/drivers/spi/spi_ll_stm32.c b/drivers/spi/spi_ll_stm32.c
index 6d15a4fc39..b81444307e 100644
--- a/drivers/spi/spi_ll_stm32.c
+++ b/drivers/spi/spi_ll_stm32.c
@@ -112,7 +112,10 @@ static void spi_stm32_shift_m(SPI_TypeDef *spi, struct spi_stm32_data *data)
        }
 
        while (!ll_func_rx_is_not_empty(spi)) {
-               /* NOP */
+               if (!LL_SPI_IsEnabled(spi)) {
+                       LOG_INF("SPI Enable");
+                       LL_SPI_Enable(spi);
+               }
        }
 
        if (SPI_WORD_SIZE_GET(data->ctx.config->operation) == 8) {
```

`BUS FAULT` occurs even at 16MHz.
```
*** Booting Zephyr OS build zephyr-v1.13.0-13521-ga22c423b115e  ***
[00:00:00.216,000] <inf> sx1276: SX1276 Version:12 found
[00:00:00.223,000] <inf> lora_send: Data sent!
[00:00:00.234,000] <inf> lora_send: Data sent!
[00:00:00.245,000] <inf> lora_send: Data sent!
[00:00:00.256,000] <inf> lora_send: Data sent!
[00:00:00.267,000] <inf> lora_send: Data sent!
[00:00:00.278,000] <inf> lora_send: Data sent!
[00:00:00.290,000] <inf> lora_send: Data sent!
[00:00:00.301,000] <inf> lora_send: Data sent!
[00:00:00.312,000] <inf> lora_send: Data sent!
[00:00:00.323,000] <inf> lora_send: Data sent!
[00:00:00.334,000] <inf> lora_send: Data sent!
[00:00:00.345,000] <inf> lora_send: Data sent!
[00:00:00.356,000] <inf> lora_send: Data sent!
[00:00:00.367,000] <inf> spi_ll_stm32: SPI Enable
[00:00:01.024,000] <err> os: ***** BUS FAULT *****
[00:00:01.024,000] <err> os:   Imprecise data bus error
[00:00:01.024,000] <err> os: r0/a1:  0x80000000  r1/a2:  0x07fee954  r2/a3:  0x00000000
[00:00:01.024,000] <err> os: r3/a4:  0x07fee954 r12/ip:  0x0000001c r14/lr:  0x0800477b
[00:00:01.024,000] <err> os:  xpsr:  0x61000000
[00:00:01.024,000] <err> os: Faulting instruction address (r15/pc): 0x080064ee
[00:00:01.024,000] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:00:01.024,000] <err> os: Current thread: 0x200004c0 (unknown)
[00:00:01.060,000] <err> os: Halting system
``

erwango, 2020-01-31:
ok, thanks.
One additional question: Is that a regression?

KwonTae-young, 2020-02-01:
@erwango 
There was no difference.

KwonTae-young, 2020-02-07:
@erwango
I tested by connecting `nucleo_l432kc` and [SX1276MB1LAS](https://os.mbed.com/components/SX1276MB1xAS/).
Symptoms were also reproduced in nucleo_l432kc.
One interesting point is that by modifying the sx1276 driver as shown below, the symptom was not reproduced.(96b_wistrio, nucleo_l432kc)
I think it might be a bug in the sx1276 driver.
``` diff
diff --git a/drivers/lora/sx1276.c b/drivers/lora/sx1276.c
index 1ef72e9417..cd484fe316 100644
--- a/drivers/lora/sx1276.c
+++ b/drivers/lora/sx1276.c
@@ -149,7 +149,7 @@ static void sx1276_irq_callback(struct device *dev,
 
        for (i = 0; i < SX1276_MAX_DIO; i++) {
                if (pin == sx1276_dio_pins[i]) {
-                       (*DioIrq[i])(NULL);
+                       //(*DioIrq[i])(NULL);
                }
        }
 }
```

CC: @Mani-Sadhasivam

erwango, 2020-02-07:
Thank for this update @KwonTae-young

erwango, 2020-02-12:
Assigning to @Mani-Sadhasivam to check LORA side.

Mani-Sadhasivam, 2020-02-12:
@KwonTae-young I can reproduce this issue. But it is random in nature as you said. I'm seeing 1/5 times and sometimes do not. But nevertheless, it is an issue. Let me dig into it soon and update this issue.

Mani-Sadhasivam, 2020-02-12:
@KwonTae-young @erwango So looks like the issue arises when two consecutive DIO1 interrupts arrive one after the other. Then the board hangs. I can confirm that the `DioIrq` callback gets executed successfully for the second time though.

Will dig more into this.

Mani-Sadhasivam, 2020-02-14:
@KwonTae-young Some observations:

1. The second DIO0 IRQ seems to be not coming from the modem. When the second IRQ arrived, I checked the `RegIrqFlags` register and it is `0x00`.  So this might be a false interrupt on the DIO0 pin. What wonders me is, the DIO0 pin has debounce enabled in pinmux so this should never happen.

2. The fact that the board hang is avoided when commenting out `(*DioIrq[i])(NULL);` is that, the DIO0 callback will never get called so the corresponding IRQ will not be cleared in the modem. So this pin will stay high and there won't be any further interrupts coming. So in short, we will only have the DIO0 interrupt happening once and it will never arise later.

Mani-Sadhasivam, 2020-02-14:
> What wonders me is, the DIO0 pin has debounce enabled in pinmux so this should never happen.

Ah, the STM32 gpio driver is not handling the debounce functionality. So this might be the culprit.

Mani-Sadhasivam, 2020-02-14:
@erwango

KwonTae-young, 2020-02-20:
I checked the DIO0 pin.
But I didn't find a special waveform to say Debounce.
[image]  
[image]

Sometimes there is a section where DIO0 stays high, but this did not lead to freezing.  
[image]

One strange thing is that I'm calling lora_send() every `10ms`, but the `DIO0(TxDone)` interrupt was happening every `300ms`.
Wouldn't it be a problem if we called lora_send() before the TxDone interrupt occurred?

I made the following change to wait for the TxDone interrupt to occur when using the LoRa API.
Could this be the solution?
I did a test and it didn't stop.
``` diff
diff --git a/drivers/lora/sx1276.c b/drivers/lora/sx1276.c
index 1ef72e9417..43c4820696 100644
--- a/drivers/lora/sx1276.c
+++ b/drivers/lora/sx1276.c
@@ -53,6 +53,7 @@ struct sx1276_data {
        RadioEvents_t sx1276_event;
        u8_t *rx_buf;
        u8_t rx_len;
+       bool tx_done;
 } dev_data;
 
 bool SX1276CheckRfFrequency(uint32_t frequency)
@@ -320,6 +321,11 @@ void SX1276SetRfTxPower(int8_t power)
 
 static int sx1276_lora_send(struct device *dev, u8_t *data, u32_t data_len)
 {
+       if (dev_data.tx_done == false) {
+               return -1;
+       }
+       dev_data.tx_done = false;
+
        Radio.SetMaxPayloadLength(MODEM_LORA, data_len);
 
        Radio.Send(data, data_len);
@@ -329,6 +335,7 @@ static int sx1276_lora_send(struct device *dev, u8_t *data, u32_t data_len)
 
 static void sx1276_tx_done(void)
 {
+       dev_data.tx_done = true;
        Radio.Sleep();
 }
 
@@ -347,6 +354,10 @@ static int sx1276_lora_recv(struct device *dev, u8_t *data, u8_t size,
 {
        int ret;
 
+       if (dev_data.tx_done == false) {
+               return -1;
+       }
+
        Radio.SetMaxPayloadLength(MODEM_LORA, 255);
        Radio.Rx(0);
 
@@ -378,6 +389,9 @@ static int sx1276_lora_recv(struct device *dev, u8_t *data, u8_t size,
 static int sx1276_lora_config(struct device *dev,
                              struct lora_modem_config *config)
 {
+       if (dev_data.tx_done == false) {
+               return -1;
+       }
 
        Radio.SetChannel(config->frequency);
 
@@ -482,6 +496,7 @@ static int sx1276_lora_init(struct device *dev)
 
[... 66 lines cut ...]
[00:00:00.608,000] <err> lora_send: LoRa send failed                                           
[00:00:00.618,000] <err> lora_send: LoRa send failed                                           
[00:00:00.628,000] <err> lora_send: LoRa send failed                                           
[00:00:00.638,000] <err> lora_send: LoRa send failed                                           
[00:00:00.648,000] <err> lora_send: LoRa send failed                                           
[00:00:00.658,000] <err> lora_send: LoRa send failed                                           
[00:00:00.668,000] <err> lora_send: LoRa send failed                                           
[00:00:00.679,000] <err> lora_send: LoRa send failed                                           
[00:00:00.689,000] <err> lora_send: LoRa send failed                                           
[00:00:00.699,000] <err> lora_send: LoRa send failed                                           
[00:00:00.709,000] <err> lora_send: LoRa send failed                                           
[00:00:00.719,000] <err> lora_send: LoRa send failed                                           
[00:00:00.729,000] <err> lora_send: LoRa send failed                                           
[00:00:00.739,000] <err> lora_send: LoRa send failed                                           
[00:00:00.749,000] <err> lora_send: LoRa send failed                                           
[00:00:00.759,000] <err> lora_send: LoRa send failed                                           
[00:00:00.769,000] <err> lora_send: LoRa send failed                                           
[00:00:00.780,000] <err> lora_send: LoRa send failed                                           
[00:00:00.790,000] <err> lora_send: LoRa send failed                                           
[00:00:00.800,000] <err> lora_send: LoRa send failed                                           
[00:00:00.810,000] <err> lora_send: LoRa send failed                                           
[00:00:00.822,000] <inf> lora_send: Data sent!
...
...
```

Mani-Sadhasivam, 2020-04-15:
Fixed by #24121

closed 2020-04-15 by Mani-Sadhasivam