zephyrproject-rtos/zephyr #22085: HCI/CCO/BV-07-C & HCI/GEV/BV-01-C failing in EDTT

aescolar, 2020-01-22:
**Describe the bug**
HCI/CCO/BV-07-C [BR/EDR Commands Not Supported on LE Device]
&
HCI/GEV/BV-01-C [Status return for Unsupported Commands]
are failing in EDTT, due to a mismatch in the expected response from the controller.
The test expects a "Command Status event" while the controller returns a “command completed” event with invalid command status.

A proposal to change the test was sent to the EDTT to align it with the controller https://github.com/EDTTool/EDTT/pull/2 , but no immediate agreement was reached about which should be the appropriate response from the controller.

**To Reproduce**
See `tests/bluetooth/bsim_bt/edtt_ble_test_app/README.txt`
And either run all tests or just 
`tests/bluetooth/bsim_bt/edtt_ble_test_app/tests_scripts/hci.sh`

**Expected behavior**
Passing tests

**Impact**
An immediate workaround is to disable the test.
But there is a significant probability that the implementation is not correct. Therefore it would be much better to clarify what is the expected behavior and either fix the controller or test.

**Environment (please complete the following information):**
 - OS: Linux (Ubuntu 18.04)
 - Toolchain: host gcc
 - Commit SHA: Today's master, but this behavior has been the same for very long.

aescolar, 2020-01-22:
CC @joerchan @carlescufi @heri-developer

joerchan, 2020-01-22:
@aescolar Which command(s)?

aescolar, 2020-01-23:
>@aescolar Which command(s)?

@joerchan (Yep, that is relevant to state :D ). Inquiry.
https://github.com/EDTTool/EDTT/blob/public_master/src/tests/hci_verification.py#L310
https://github.com/EDTTool/EDTT/blob/public_master/src/components/basic_commands.py#L393

Vinayak's fix proposal was this one: https://github.com/EDTTool/EDTT/pull/2

joerchan, 2020-02-14:
The pass verdict of the test is quite clear and the controller should be passing:

Pass Verdict
The IUT returns an HCI Command Complete or HCI Command Status Event with Status =
Unknown_HCI_Command.

@aescolar Please update EDTT to accept both responses.

PR #22867 (2020-02-17): Bluetooth: EDTT: Handle unknown response
Related to: https://github.com/EDTTool/EDTT/pull/11
I'm not sure how to handle the west manifest with the zephyr mirror.

Update HCI test application to give the same unsupported command RSP from the command complete and command status HCI events.

Enabled the tests that now are able to pass.

Remove deprecated defines and added some debug tracing that I had been using.

Fixes: #22085
closed 2020-02-17 by jhedberg