zephyrproject-rtos/zephyr #22088: Bluetooth Mesh friendship is cleared due to no Friend response reception

DHortelano, 2020-01-22:
**Describe the bug**
In the friendship established between a Friend Node (FN) and a Low Power Node (LPN) in a Bluetooth Mesh, the LPN generates and sends the Friend Poll message, which is correctly received by the FN. However, after that, the LPN is not able to start the scanning, with the following debug output:
[00:00:32.176,910] <err> bt_mesh_adv: starting scan failed (err -5)   
This produces the FN response is not received, and after some retries the friendship is cleared.

**To Reproduce**
Steps to reproduce the behavior:
1. Establish a Friendiship between a LPN and a FN, using default configuration
2. After the Friend Poll, the LPN is not able to scan 
3. Output shows <err> bt_mesh_adv: starting scan failed (err -5)  
4. After the timeout expires, the friendship is cleared

**Expected behavior**
The LPN receives the Friend response 

**Screenshots or console output**
Debugging output from LPN using CONFIG_BT_MESH_DEBUG_LOW_POWER and CONFIG_BT_MESH_DEBUG_ADV options:
```
Initializing...                                                                                                                                  
Unicast address: 0x0003                                                                                                                          
Bluetooth initialized                                                                                                                            
Mesh initialized                                                                                                                                 
Loading stored settings                                                                                                                          
Provisioning completed                                                                                                                           
Configuring...                                                                                                                                   
Publishing heartbeat messages                                                                                                                    
Configuration complete                                                                                                                           
[00:00:00.011,169] <inf> fs_nvs: 8 Sectors of 4096 bytes                                                                                         
[00:00:00.011,169] <inf> fs_nvs: alloc wra: 0, ff0                                                                                               
[00:00:00.011,169] <inf> fs_nvs: data wra: 0, 0                                                                                                  
[00:00:00.012,512] <inf> bt_hci_core: HW Platform: Nordic Semiconductor (0x0002)                                                                 
[00:00:00.012,542] <inf> bt_hci_core: HW Variant: nRF52x (0x0002)                                                                                
--- 9 messages dropped ---                                                                                                                       
[00:00:00.024,627] <dbg> bt_mesh_lpn.bt_mesh_lpn_init:                                                                                           
[00:00:00.024,627] <dbg> bt_mesh_adv.bt_mesh_scan_enable:                                                                                        
[00:00:00.024,902] <dbg> bt_mesh_lpn.bt_mesh_lpn_init: Waiting 1000 ms for messages                                                              
[00:00:00.024,902] <dbg> bt_mesh_lpn.lpn_set_state: disabled -> timer                                                                            
[00:00:00.067,260] <dbg> bt_mesh_lpn.bt_mesh_lpn_msg_received: Restarting establishment timer                                                    
[00:00:00.069,091] <dbg> bt_mesh_lpn.bt_mesh_lpn_msg_received: Restarting establishment timer                                                    
[00:00:00.069,549] <dbg> bt_mesh_lpn.bt_mesh_lpn_msg_received: Restarting establishment timer                                                    
[00:00:00.069,976] <dbg> bt_mesh_lpn.bt_mesh_lpn_msg_received: Restarting establishment timer                                                    
[00:00:00.070,190] <dbg> bt_mesh_lpn.bt_mesh_lpn_group_add: group 0xc000                                                                         
[00:00:00.070,434] <dbg> bt_mesh_lpn.bt_mesh_lpn_msg_received: Restarting establishment timer                                                    
[00:00:00.070,892] <dbg> bt_mesh_lpn.bt_mesh_lpn_msg_received: Restarting establishment timer                                                    
[00:00:00.071,044] <dbg> bt_mesh_lpn.bt_mesh_lpn_msg_received: Restarting establishment timer                                                    
[00:00:00.071,289] <wrn> bt_mesh_cfg_cli: Unexpected App Key Status message                                                                      
[00:00:00.071,289] <dbg> bt_mesh_lpn.bt_mesh_lpn_msg_received: Restarting establishment timer                                                    
[00:00:00.071,533] <wrn> bt_mesh_cfg_cli: Unexpected Model App Status message                                                                    
[00:00:00.071,533] <dbg> bt_mesh_lpn.bt_mesh_lpn_msg_received: Restarting establishment timer                                                    
[00:00:00.071,746] <wrn> bt_mesh_cfg_cli: Unexpected Model App Status message                                                                    
[00:00:00.071,777] <dbg> bt_mesh_lpn.bt_mesh_lpn_msg_received: Restarting establishment timer                                                    
[00:00:00.072,021] <wrn> bt_mesh_cfg_cli: Unexpected Model Subscription Status message                                                           
[00:00:00.072,021] <dbg> bt_mesh_lpn.bt_mesh_lpn_msg_received: Restarting establishment timer                                                    
[00:00:00.072,265] <wrn> bt_mesh_cfg_cli: Unexpected Heartbeat Publication Status message                                                        
[00:00:00.072,296] <dbg> bt_mesh_lpn.bt_mesh_lpn_msg_received: Restarting establishment timer                                                    
[00:00:00.072,601] <wrn> bt_mesh_transport: Ignoring heartbeat to non-subscribed destination                                                     
[00:00:00.072,631] <dbg> bt_mesh_lpn.bt_mesh_lpn_msg_received: Restarting establishment timer                                                    
[00:00:00.072,784] <dbg> bt_mesh_adv.bt_mesh_adv_send: type 0x01 len 21: 373fa347ad8cf0510fa9e4e308e7016~                                        
[00:00:00.072,875] <dbg> bt_mesh_adv.adv_send: type 1 len 21: 373fa347ad8cf0510fa9e4e308e7016~                                                   
[00:00:00.072,875] <dbg> bt_mesh_adv.adv_send: count 3 interval 20ms duration 120ms                                                              
[00:00:00.073,577] <dbg> bt_mesh_adv.adv_send: Advertising started. Sleeping 120 ms                                                              
[00:00:00.193,786] <dbg> bt_mesh_adv.adv_send: Advertising stopped                                                                               
[00:00:01.072,692] <dbg> bt_mesh_lpn.lpn_timeout: state: timer                                                                                   
[00:00:01.072,692] <dbg> bt_mesh_lpn.lpn_timeout: Starting to look for Friend nodes                                                              
[00:00:01.072,723] <dbg> bt_mesh_lpn.lpn_set_state: timer -> enabled                                                                             
[00:00:01.072,723] <dbg> bt_mesh_lpn.send_friend_req:                                                                                            
[00:00:01.072,875] <dbg> bt_mesh_adv.bt_mesh_adv_send: type 0x01 len 28: 37d544ee68f31aa29525ffd3c210726~                                        
[00:00:01.072,937] <dbg> bt_mesh_adv.adv_send: type 1 len 28: 37d544ee68f31aa29525ffd3c210726~                                                   
[00:00:01.072,937] <dbg> bt_mesh_adv.adv_send: count 2 interval 20ms duration 90ms                                                               
[... cut ...]
[00:00:09.415,161] <dbg> bt_mesh_adv.bt_mesh_scan_disable:                                                                                       
[00:00:09.415,405] <dbg> bt_mesh_lpn.lpn_set_state: wait offer -> established                                                                    
```

**Environment (please complete the following information):**
 - Zephyr project in nRF52840 boards

**Additional context**
Tracking the error, it is generated, apparently, in hci_core.c, when the function "start_le_scan(u8_t scan_type, u16_t interval, u16_t window)"  uses the "err = set_random_address(&bt_dev.id_addr[0].a);" function.

aescolar, 2020-01-22:
CC @jhedberg

aurel32, 2020-01-22:
@DHortelano, what configuration values do you use for `BT_MESH_LPN_RECV_DELAY` and `BT_MESH_LPN_SCAN_LATENCY`. To get it working on a nRF52 device, I had to increase `BT_MESH_LPN_RECV_DELAY` to 150 and keep `BT_MESH_LPN_SCAN_LATENCY` to the default 10.

I am not sure what should be the exact values, it seems that the error happens when the scanning phase starts too close to the end of the advertisement phase, although I haven't really debugged more than testing different values.

jhedberg, 2020-01-22:
CC @trond-snekvik

DHortelano, 2020-01-22:
Hi @aurel32, I used ```BT_MESH_LPN_RECV_DELAY=100``` and ```CONFIG_BT_MESH_LPN_SCAN_LATENCY=10```. I have tried to increase the ```BT_MESH_LPN_RECV_DELAY``` to 150, and you are right, it works fine, thank you!

> it seems that the error happens when the scanning phase starts too close to the end of the advertisement phase,

I completely agree with you, and I think it could be due to the change of address type when the scan starts. I have added a check to do that only when the ```scan_type``` is ```BT_HCI_LE_SCAN_ACTIVE```. This also works, and it allows to use a shorter ```BT_MESH_LPN_RECV_DELAY``` (I tried 50). However, I don't know if this modification is allowed by the Bluetooth Mesh standard...

jhedberg, 2020-01-22:
It sounds like this is all related to the fact that we don't have super precise control of advertising timing due to our use of standard HCI (and legacy 4.0 HCI at that). This should all hopefully get fixed once we start using Mesh vendor extensions for HCI. Meanwhile, the only thing I can think of improving is to change the default values of these Kconfig options.

trond-snekvik, 2020-01-29:
Looks like this is just a symptom of a bug in hci_core. I made a PR for skipping the address setting when using a passive scanner, like in the mesh, as it's not used. This should remove the call generating the error. @joerchan plans on fixing the overarching problem in his AE work.

@DHortelano #22313 should fix the problem, even for your original timing parameters.

joerchan, 2020-01-29:
> @joerchan plans on fixing the overarching problem in his AE work.

To clarify, this problem can be avoided if the controller supports AE :) It is still something we have to deal with when AE commands are not available.

PR #22313 (2020-02-03): Bluetooth: Skip address setting for passive scan
The passive scanner isn't actually using the advertising address, and
doesn't need to set it during scan initialization.

set_random_addr always fails if there's an active advertiser, which
would cause needless complications for the Bluetooth Mesh LPN case.

Fixes #22088.
commit a112f156 (2020-02-03): Bluetooth: host: Skip set passive scan when not scanning with identity

When privacy is disabled by default the scanner still protects it's
identity with the use of NRPA addresses. We should not set the identity
address for the passive scanner unless the Kconfig option to scan with
the identity has been enabled.

This makes passive scanner behave the same way as an active scanner
since none of them will report directed advertising reports towards
the scanners identity.

This also enables the advertiser to switch out the random address which
is needed for the Bluetooth Mesh LPN case.

Fixes #22 [... cut]
closed 2020-02-03 by jhedberg