zephyrproject-rtos/zephyr #22091: Blink-Led example doesn't build on Nucleo_L476RG, STM32F4_DISCOVERY, Nucleo_F302R8, Nucleo_F401RE

giellamoswhard, 2020-01-22:
**Describe the bug**
I've tried to build the BlinkLed example with the boards o_L476RG, STM32F4_DISCOVERY, Nucleo_F302R8, Nucleo_F401RE but it doesn't build 
#error "Choose supported PWM driver" 
**To Reproduce**
Steps to reproduce the behavior:
1. cd zephyrproject/zephyr/samples/basic/blink_led
2. west build -b nucleo_f401re -- -DBOARD_ROOT=/home/giellamo/projects/zephyrproject/zephyr/boards
3. ../src/main.c:28:2: error: #error "Choose supported PWM driver"
 #error "Choose supported PWM driver"
  ^~~~~
../src/main.c: In function 'main':
../src/main.c:46:31: error: 'PWM_DRIVER' undeclared (first use in this function); did you mean 'PWR_CR_VOS'?
  pwm_dev = device_get_binding(PWM_DRIVER);
                               ^~~~~~~~~~
                               PWR_CR_VOS

**Expected behavior**
It should build

**Environment (please complete the following information):**
 - OS: .g. Linux, MacOS, Windows)
 - Toolchain Zephyr SDK
 - Commit 2.1.99

**Additional context**
The boards seems to miss the pwmleds in the dts

erwango, 2020-01-22:
Boards supporting/enabling PWM should declare the following or equivalent:
```
	pwmleds {
		compatible = "pwm-leds";		red_pwm_led: red_pwm_led {
			pwms = <&{/soc/timers@40014000/pwm} 1 4>;
		};
	};
	aliases {
        ...
		pwm-led0 = &red_pwm_led;
		...
	};
```

sercantun, 2020-01-23:
Could you please explain what is the 1 and 4 arguments are used for ?

erwango, 2020-01-24:
@sercantun, information is available in `dts/bindings/pwm/st,stm32-pwm.yaml`:
```
pwm-cells:
  - channel
# period in terms of nanoseconds
  - period
```

erwango, 2020-02-11:
@giellamoswhard can you check #22277 answers your question ?

giellamoswhard, 2020-02-11:
Yes. Thank you very much.

erwango, 2020-02-12:
I'm reopening the point.
It will be closed automatically when #22777 gets merged

closed 2020-08-12 by github-actions[bot]