zephyrproject-rtos/zephyr #22128: frdm_k82f:samples/drivers/spi_fujitsu_fram/sample.drivers.spi.fujitsu_fram fails

pabigot, 2020-01-23:
`sanitycheck -p frdm_k82f -T samples/drivers/spi_fujitsu_fram` fails on master, blocking successful shippable in #21776.  

```
/mnt/nordic/zp/zephyr/samples/drivers/spi_fujitsu_fram/src/main.c: In function 'main':
/mnt/nordic/zp/zephyr/samples/drivers/spi_fujitsu_fram/src/main.c:146:27: error: 'DT_SPI_1_NAME' undeclared (first use in this function); did you mean 'DT_SIM_NAME'?
  spi = device_get_binding(DT_SPI_1_NAME);
                           ^~~~~~~~~~~~~
                           DT_SIM_NAME
```

It appears a legacy devicetree override is being used here that is not defined for this platform.  It's not clear where to add it or what its value should be.

Alternatively this could be switched to use `DT_ALIAS_SPI_1_LABEL`, at the risk of breaking some platform that doesn't provide that as a label.

Alternatively this sample could be converted to devicetree so the proper bus could located automatically.

Alternatively this sample could be removed, since it isn't demonstrating a driver but a hand-coded API for a single chip.

Alternatively....

galak, 2020-01-23:
Adding @MaureenHelm as not sure if this might be related to the NXP DT cleanup she did.

MaureenHelm, 2020-01-23:
It is indeed related to my dts cleanup, but strange that it took this long to fail in master. The patch was merged on 15 Jan.

I vote for removing the sample. It appears to have been added for the Intel quark board, which is no longer supported. I don't know of any other boards that have this particular spi slave device. @nashif, is that ok with you?

nashif, 2020-01-23:
> It appears to have been added for the Intel quark board, which is no longer supported.

this is not related to intel quark boards. and can be attached to board. This module which is available as a breakout from adafruit and others was used to test SPI, afaik there is also an i2c variant:

- https://www.adafruit.com/product/1897
- https://www.adafruit.com/product/1895

can be fixed now with:
```
diff --git a/samples/drivers/spi_fujitsu_fram/sample.yaml b/samples/drivers/spi_fujitsu_fram/sample.yaml
index e1bae050fd..25371729a0 100644
--- a/samples/drivers/spi_fujitsu_fram/sample.yaml
+++ b/samples/drivers/spi_fujitsu_fram/sample.yaml
@@ -5,4 +5,4 @@ tests:
     tags: drivers fram
     harness: fram
     depends_on: gpio spi
-    filter: CONFIG_SPI_1
+    filter: dt_alias_exists("spi-1")
diff --git a/samples/drivers/spi_fujitsu_fram/src/main.c b/samples/drivers/spi_fujitsu_fram/src/main.c
index fecbd83ec0..2e666c707d 100644
--- a/samples/drivers/spi_fujitsu_fram/src/main.c
+++ b/samples/drivers/spi_fujitsu_fram/src/main.c
@@ -143,7 +143,7 @@ void main(void)

        printk("fujitsu FRAM example application\n");

-       spi = device_get_binding(DT_SPI_1_NAME);
+       spi = device_get_binding(DT_ALIAS_SPI_1_LABEL);
        if (!spi) {
                printk("Could not find SPI driver\n");
                return;
```
for example, but we should move this to tests/ and clean it up a bit and probably create a real driver for that device. Btw, for the test shield we were planning to use something similar.

nashif, 2020-01-23:
> It is indeed related to my dts cleanup, but strange that it took this long to fail in master. The patch was merged on 15 Jan.

it was failing in master since then, see https://app.shippable.com/github/zephyrproject-rtos/ci-test/runs/3359/19/console

PR #22138 (2020-01-23): samples: drivers: fix build of spi_fujitsu_fram sample
Use DTS filtering and get binding based on DTS variable.

Fixes #22128
closed 2020-01-23 by nashif