zephyrproject-rtos/zephyr #22140: Exiting deep sleep without button help; nrf52832

Mhrooczny, 2020-01-23:
**Describe the bug**
Im using sample called `system_off` on nrf52832 mcu. It is any possibility to return from `SYS_POWER_STATE_DEEP_SLEEP_1` mode not using button? I wanted to create low power app and that `SYS_POWER_STATE_DEEP_SLEEP_1` is very satisfying. I tried to use functions:
- _sys_resume_from_deep_sleep()
- _sys_resume();
- sys_set_power_state() / sys_pm_force_power_state with different argument's:
SYS_POWER_STATE_AUTO / SYS_POWER_STATE_ACTIVE / SYS_POWER_STATE_MAX
but any combination of above wont restore context to the point where system entered the deep sleep state.

**To Reproduce** 
1. west build -b nrf52_pca10040  samples/boards/nrf52/system_off
2. west flash

**Environment:**
 - OS: Linux
 - Zephyr 2.1.0

carlescufi, 2020-01-23:
I don’t think this is possible without external hardware that triggers a pin. @anangl or @pabigot can confirm

anangl, 2020-01-24:
It's not possible. As stated in [documentation](https://infocenter.nordicsemi.com/topic/com.nordic.infocenter.nrf52832.ps.v1.1/power.html?cp=4_2_0_17_1#unique_1707892264), to wake up the device from System OFF, you need of the following signals:

> 1. The DETECT signal, optionally generated by the GPIO peripheral
> 2. The ANADETECT signal, optionally generated by the LPCOMP module
> 3. The SENSE signal, optionally generated by the NFC module to “wake-on-field”
> 4. A reset

Thus, you'll need some external hardware to generate one of these.

Mhrooczny, 2020-01-24:
Thank you for answer and help.  Therefore if i wanted to avoid using external hardware and build low power app i can only use k_sleep function as `sleep state` to get lowest power consumption?

pabigot, 2020-01-24:
If I understand what you're doing correctly: use `CONFIG_SYS_POWER_DEEP_SLEEP_STATES=n` to prevent `k_sleep()` from ever turning the MCU off (which should be the default), then any time the application is doing nothing it should be in the sleep state corresponding to `k_sleep()`.  You can use any function that causes a sleep to make a thread inactive, e.g. `k_poll()` if you want to wake up on some event that you can poll for.

closed 2020-09-19 by github-actions[bot]