zephyrproject-rtos/zephyr #22151: hal_nordic: nrfx: doxygen: Reference to missing nrfx/templates

ulfalizer, 2020-01-24:
(GitHub Issues are disabled on the [hal-nordic](https://github.com/zephyrproject-rtos/hal_nordic) repo. Piggybacking here instead for now.)

[nrfx/doc/nrfx.doxyfile](https://github.com/zephyrproject-rtos/hal_nordic/blob/a1f24a4ee9774fe96390f2c6473b490c88b01ef6/nrfx/doc/nrfx.doxyfile#L798) has `../templates` in `INPUT`, but no `nrfx/templates/` directory exists.

I'm guessing the `@nrfx_glue` and `@nrfx_log` tags are supposed to come from there. They end up missing now.

Might be nice to have a README for the repository. It's a bit hard to see how all the different files fit together now. There's `nrfx_glue.c` and `nrfx_glue.h` at the top level too, for example.

ulfalizer, 2020-01-24:
CC @anangl @rlubos @carlescufi

anangl, 2020-01-24:
> [nrfx/doc/nrfx.doxyfile](https://github.com/zephyrproject-rtos/hal_nordic/blob/a1f24a4ee9774fe96390f2c6473b490c88b01ef6/nrfx/doc/nrfx.doxyfile#L798) has `../templates` in `INPUT`, but no [...]

Yes. Also `doc/main_page.dox` [references](https://github.com/zephyrproject-rtos/hal_nordic/blob/a1f24a4ee9774fe96390f2c6473b490c88b01ef6/nrfx/doc/main_page.dox#L28) this directory, for the second time [here](https://github.com/zephyrproject-rtos/hal_nordic/blob/a1f24a4ee9774fe96390f2c6473b490c88b01ef6/nrfx/doc/main_page.dox#L74). I think that this directory should be added to the hal_nordic repo. [Here](https://github.com/NordicSemiconductor/nrfx/tree/master/templates) is what it contains in the original nrfx repo.

ulfalizer, 2020-01-24:
Were those files accidentally put at the top level when they were supposed to go in `nrfx/templates`, or was that deliberate?

anangl, 2020-01-28:
@ulfalizer The files (`nrfx_glue.h`, `nrfx_log.h`, and `nrfx_config.h`) at the top level are needed for integration of nrfx into the environment of Zephyr. And the ones in `nrfx/templates` are just templates of those. They are provided in the nrfx repository as a starting point for users that want to do their own integration of nrfx, and they are supposed to be filled with the content required by the environment that nrfx is integrated into. This is described in [main_page](https://github.com/NordicSemiconductor/nrfx/blob/master/doc/main_page.dox#L15).

anangl, 2020-07-29:
Addressed in https://github.com/zephyrproject-rtos/hal_nordic/pull/13.

closed 2020-07-29 by anangl