zephyrproject-rtos/zephyr #22158: flash_img: support arbitrary flash devices

hakonfam, 2020-01-24:
In a multi SoC system (A and B), it is often necessary for A to download a new firmware for B. A will store this to some external flash, verify its validity once the whole file is in place in external flash, and then write it to B in some manner.

There are several issues with solutions like this, and `flash_img.c` solves several of these. Most importantly it handles the buffered writes.

Secondly, `flash_img.c` provides the `flash_img_context` struct, which can easily be saved using the `settings` library. This allows A to be reset during the download of the B firmware, and then resume the download exactly where it left off. Note that data transfer is very expensive in some cases, and the firmware can be huge.

My proposal is that we generalize the functionality found in `flash_img.c` so that data can be written to any flash area inside any flash device. One instantiation of this generic library will be the current `flash_img.c`.

This can be abstracted into a "write huge file to flash" library, which when coupled with the `settings` library storing the `flash_img context` will allow resumption of the fetching of these huge files.

Storing the flash_img context to flash through the `settings` library should be configurable.

hakonfam, 2020-01-24:
I assigned you @nvlsianpu mostly to get your feedback on whether or not this sounds like a reasonable proposal :)

nvlsianpu, 2020-01-24:
> My proposal is that we generalize the functionality found in flash_img.c so that data can be written to any flash area inside any flash device. One instantiation of this generic library will be the current flash_img.c.
.OK

> Secondly, flash_img.c provides the flash_img_context struct, which can easily be saved using the settings library. This allows A to be reset during the download of the B firmware, and then resume [...]

Settings can be used or other storage solution (like naked FS). Maybe application can fetch relevant flash_img_context from storage as it does today? Or you want to integrate this into the module.

hakonfam, 2020-01-24:
> Settings can be used or other storage solution (like naked FS). Maybe application can fetch relevant flash_img_context from storage as it does today? Or you want to integrate this into the module.

This should be outside the module IMO, to make it as simple as possible. But it is the existence struct like `flash_img_context` that is important, so that storing the process and resuming is easy.

Laczen, 2020-01-28:
@hakonfam, I like the idea of trying to avoid restarting a download. Using the settings module to store flash_img_context is not a good idea: every write to flash_img_context will need to be written to settings and the settings storage area will wear out (very fast) as a result of this.

nvlsianpu, 2020-01-28:
^^ @hakonfam Are you going to write context on every firmware chunk downloaded?

hakonfam, 2020-01-28:
Yes. One can, however, use a settings backend which distributes the writes across a large area, to minimize the flash wear. 

This is something the user must choose. In some use cases it is not acceptable to restart fw downloads from start, but spending a lot of flash in avoiding flash wear is acceptable.

Laczen, 2020-01-28:
@hakonfam, please do a little calculation on how large the settings flash area should be to get an acceptable lifetime and number of upgrades.

I agree with the requirement, the solution is not OK. In the case where communication is problematic, some extra communication between the sender and the receiver is required. The receiver should be able to request a restart from a certain location. In that case nothing needs to be stored in settings (or maybe just the last OK written offset), and the fw download can restart properly.

Another solution might be to store the fw image completely in the settings backend, but then it would be required to start the update from the settings backend.

hakonfam, 2020-01-28:
What if the interval at which the context should be stored in flash would be configurable by the user. Then, depending on the nature of the application, you might end up with everything from 2 writes per upgrade, to one write per received firmware fragment.

Laczen, 2020-01-28:
> Edit:
> 
> Regarding you comment on reporting back the offset where the download should continue, I agree that we can assume that this is in place.

If this is in place (**it is not at the moment**) I see no problem in the solution, it is not required to store the context but only the end of what was written to flash. It could be decided only to store erased sector information and to restart with a new erase of this sector. It should then only be required to store the "number of sector writes" in the settings backend. The number of writes is very limited and the influence is negligible on settings flash wear.

closed 2020-03-10 by jhedberg