zephyrproject-rtos/zephyr #22161: add shell comand for the settings subsystem

nvlsianpu, 2020-01-24:
**settings shell commands**
Add bunch of commands which allows to : 
- write a key-value pair
- read a key-value pair 
- delete a key-value pair

Commands shall support basic C types encoding in selected endianess, string and binary blobs.

**Objective**
Mechanism which allows to create provisioning flash raw image of the settings storage which can be used to prepare storage content for the manufacturer provisioning using natvie_posix targets and flash_simulator (on flash device file).

nvlsianpu, 2020-01-24:
cc @carlescufi @nordic-krch This should be easy to be done. Mainly want to see interest and comment of the idea.

carlescufi, 2020-01-24:
@Laczen @fnde-ot @nordic-krch @de-nordic any thoughts on this?

Laczen, 2020-01-24:
I don't have an opinion on this. As long as there is no breaking of security for e.g. mesh devices this could be created. If settings could be used to break security, settings should be removed immediately.

Laczen, 2020-01-25:
I have given this some more thinking:
a. I don't see the relation between manufacturing provisioning using native_posix (in flash.bin file) and the use of shell commands.
b. This is a potential security leak: reading provisioning information through the shell.

So I think this is a bad idea. Even more, at the moment when littlefs is used to store settings there is the possibility to access this littlefs fs over newtmgr. All provisioning data can then be read over newtmgr. This possibility should be removed immediately as this breaks the mesh security.

fnde-ot, 2020-01-27:
I think would be a nice addition.

@Laczen, I thought the shell was only used for testing, but if it has some security implications for real-world applications, couldn't the settings commands simply be compiled out using some Kconfig option?

de-nordic, 2020-01-27:
Maybe implement lock command that will lock ability to modify and/or read settings from shell.
This way you may setup device via shell and then remove, via the same shell, that ability.

nvlsianpu, 2020-01-27:
I would like to make it clear:
feature I suggested in this issue is for testing, evaluation and manufacturer support. Settings shell shouldn't land in the real application (unless someone really knows what is doing). So I agree with comments regards security issue it can open.

Use case I see is the generation of storage content which can be merged with compiler output (e.g. zephyr.bin or zephyr.hex). As the result merged file can be used to flash real devices in the factory.

Above is possible as native_posix supports the flash_simulator emulated flash driver, which generates flash driver file which contains storage image. Thanks to that it is possible to generate any storage content on the developer PC using zephyr RTOS code-base.

Laczen, 2020-01-27:
@fnde-ot, @de-nordic, yes it would be possible to limit this to testing (would make it useless to setup a device) or to create a lock (but you would need to unlock it also). However both are not in line with the security for bluetooth mesh. In BT-mesh you have several keys (netkey, appkey, ...) and the knowledge of these keys is defining what you can or cannot do. If you can bypass this in any non-secure way the BT-mesh security is gone.

There are other ways to achieve the objective. If the setup is done by creating a file with setup values and this file is written to the device during the preconditioning phase (in house) there is no security problem. If preconditioning in place is required a encrypted upload of this file could be done. As long as the creation of the preconditioning file is done at a place where the keys are known you are not violating the security (this is not foreseen by the BT-mesh specifications, but probably will be added
some time in future).

Laczen, 2020-01-27:
> I would like to make it clear:
> feature I suggested in this issue is for testing, evaluation and manufacturer support. Settings shell shouldn't land in the real application (unless someone really [...]

You don't need any shell support for this.

fnde-ot, 2020-01-27:
I would actually use some settings shell support for development and testing if it were implemented. It makes it easier if one wants to play around with settings, rather than making a dedicated application.

nvlsianpu, 2020-01-27:
@Laczen Yes it is possible to be done right now without shell. But for every content I want to generate I need to write/modify a simple C application. Shell will move this modification to a shell script on PC, which will make thing easier (my subjective feeling).

Laczen, 2020-01-27:
> I would actually use some settings shell support for development and testing if it were implemented. It makes it easier if one wants to play around with settings, rather than making a dedicated application.

I agree, that for testing and development it would be an added value (not for preconditioning). When settings was first ported it had this possibility. In order to make it safe it is possible to use the following strategy:
a. Use the export function for each setting to determine if a setting can be changed: if the export returns an error code the value cannot be changed.
b. Use the set method to update in case a. returned something valid.

This would result in the designer deciding what values can be changed and what values cannot be changed. The designer could decide to allow changing only during testing by making the export function return an error code if there is no test performed.

Laczen, 2020-01-27:
> @Laczen Yes it is possible to be done right now without shell. But for every content I want to generate I need to write/modify a simple C application. Shell will move this modification to a shell script on PC, which will make thing easier (my subjective feeling).

This is exactly the bad idea I was thinking about, creating a security hole to make things easier.

carlescufi, 2020-04-07:
@nvlsianpu can we decouple the addition of commands to add/edit/remove key-value pairs as shell commands with the creation of pre-baked images? We could start by adding the shell commands and see about the images later.

nvlsianpu, 2020-04-08:
@carlescufi Yes we can decouple these ideas. Moreover I remember @Laczen example native_posix application which is designed for creation of pre-backed image. https://github.com/Laczen/zephyrmodules/tree/master/utils/native_posix/settings_preload

pfalcon, 2020-05-18:
Label "area: POSIX" is for Zephyr POSIX subsystem, not for native_posix target (confusion between the too is long-standing; the only common thing between them is that they make each other's life more complicated (this include assigning wrong ticket labels ;-) ). Removing that label. The right label is "area: native port".

Delsian, 2020-12-10:
Vote for this feature

farhangj, 2021-04-23:
I also vote for this feature. Currently having to develop a cmdline-like module to set/get params for a module that will interface with a PC program. If this feature was developed it would fit right in. I think many embedded applications use a command line to get/set params. Though it's good to keep in mind not all params are "settings"/non-volatile related.. so ideally it would allow some commands to be routed to settings and some to be routed to rest of the application. e.g. get voltage.battery is reporting a real-time value that would not come from settings. but get config.mode would be routed to settings. 
Thoughts?

rettichschnidi, 2022-05-05:
I am also in need of a shell module to read and write settings.

Thankfully, @Damian-Nordic already added the functionality to list keys and read their values. Is there any reasons (other than it being more work) why the writing part has been left out? Any reason that would prevent someone from upstreaming this feature?

Damian-Nordic, 2022-05-05:
@rettichschnidi The only reason was that I needed only listing and reading settings for test/verification purposes and I wanted to achieve that with minimum effort. Also, I wasn't sure if writing would be equally useful, but if it would it can be added for sure :).

Damian-Nordic, 2022-05-07:
@rettichschnidi Actually, I've been recently working on some settings-related stuff and needed a write command for testing, so created a PR https://github.com/zephyrproject-rtos/zephyr/pull/45459 with both write and delete commands.

PR #45459 (2022-05-13): settings: add shell commands to write and delete settings
Existing settings shell allows for listing and reading
settings only. Add the following two commands:

```
settings delete <key>
settings write <key> <value-encoded-in-hex>
```
Note: the max numbers of bytes that can be written is limited by both SETTINGS_MAX_VAL_LEN and CONFIG_SHELL_CMD_BUFF_SIZE.

Fixes #22161
closed 2022-05-13 by MaureenHelm