zephyrproject-rtos/zephyr #22185: expose thread-local data to user mode without system calls

andrewboie, 2020-01-24:
**Is your enhancement proposal related to a problem? Please describe.**
Right now, it is not possible for a user thread to access thread-local data without making a system call. The thread-local data itself is stored in a read/write area in the thread's stack object, but we don't have a way know its address without reading `_current->userspace_local_data`.

Currently we only store errno in this area, but we could add:

* The ID of the currently running thread `k_current_get()`. This blocks #15138. If we solve obtaining the thread-local storage pointer without syscalls, the ID could just be put there at thread creation, it's never going to change.

* Whether the current thread is preemptible `k_is_preempt_thread()`; the kernel would update it if the thread's priority was ever changed.

* thread's custom data struct `k_thread_custom_data_set()` `k_thread_custom_data_get()`

* a thread's string name `k_thread_name_set()` `k_thread_name_get()` `k_thread_name_copy()` (any usage from kernel mode would need to be hardened not to trust its contents)

* Ultimately, full support for GCC thread-local storage. We know everything at build time, so it's a matter of reserving enough memory in the stack object at thread creation to hold it all, likely using linker variables created directly or indirectly by the compiler.

**Solution requirement**

Implementing this will be architecture-specific, and depend on the type/capabilities of the memory protection hardware, and/or features of the CPU. Ultimately we'd have an arch_ API to read/write data in TLS at some offset from the beginning of the struct. Perhaps something similar to:

```
void arch_tls_read(void *dst, off_t offset, size_t size);
void arch_tls_write(void *src, off_t offset, size_t size);
```

This would be wrapped with friendlier macros, but the requirement would be that these could be called from user mode with no privilege elevation required. Any mistakes or malfeasance would only affect the thread's own stack or cause a page/MPU fault.

**x86 solution**

On x86 this is easy with segmentation. For example, with x86-64, on every context switch, program `FS_BASE` to the base address of the TLS area in the incoming thread's stack object. The above arch APIs would be memory references to `%fs:offset`.

**solution for other arches**

On other arches, this seems much harder, since the CPU is very likely not essentially giving us an extra register (FS_BASE) to store the base address of the thread's TLS area. The solution used on Linux builds a table in the userland address space upon first thread creation, but this doesn't map to our memory domain access schemes.

I had an idea of creating an extra MPU or MMU region at a fixed address which contains the TLS pointer, reprogrammed at every context switch. On a uniprocessor system, this could be a very small region (the minimal MPU region size, or an MMU page) that gets updated by the kernel on every context switch. The user thread could then read it to find out the address when it wants to read/write TLS data.

However, this gets harder if we want to support SMP:
- The region would have to contain the TLS pointers for all threads in a table
- The region would have to be read-only to user mode; although it's fine if a user thread stomps on its own TLS pointer, it cannot be allowed to do this for threads on other CPUs. This could be a little painful if the memory management hardware doesn't support regions that are read/write to the kernel but read-only to user mode
- The thread would have to know what CPU it is running on, without making a system call, to properly offset in the table!

**Describe alternatives you've considered**
On x86, we know what to do.

Very open to suggestions on other platforms.

**Additional context**

Further reading:
https://chao-tic.github.io/blog/2018/12/25/tls
https://www.akkadia.org/drepper/tls.pdf

andrewboie, 2020-01-24:
Note: there currently doesn't seem to be any supported platforms in Zephyr which have both SMP and an MPU, looking more closely at ARC HS that seems to have an MMU in it.

If we were to support such a system, *and* the MPU region registers could at least be *read* from user mode, we could store the TLS pointer region not at a fixed address, but have different structures for it on a per-cpu basis, and just read the base address from the MPU.

Either way, seems like we could just punt on this configuration.

andrewboie, 2020-08-27:
A lot of the work for this was already done in https://github.com/zephyrproject-rtos/zephyr/pull/26545 which we could separate out and implement for x86 and ARC

keith-packard, 2020-09-21:
Ok, I spent a bit of time today researching how this could work on x86. Of course, x86 is the oddest of odd-ball chips...

TLS support on x86 uses the GS segment register; TLS variables are referenced as small offsets from the base of the GS segment. That means we need to program the GS register correctly. And that means creating a suitable GDT entry, because the GS register references a descriptor table.

An approach that I think will work is to define CONFIG_MP_NUM_CPUS additional GDT entries and then select one of these for each CPU in the system, setting the GS in each CPU appropriately when the CPU is initialized. When switching threads, the TLS base address (stored in the thread structure) would then be used to compute a new GDT entry value which would be stored in the relevant GDT value using the current GS value.

The rest of the TLS code as provided in the picolibc patch should "just work", with the TLS segment getting allocated out of the thread stack and initialized by using data out of the executable.

all of this depends on using -ftls-model=local-exec, which is the simplest TLS model.

andrewboie, 2020-09-24:
Some reading material for x86 here https://wiki.osdev.org/Thread_Local_Storage

There's some slight differences between 32-bit and 64-bit.

32-bit we create all our GDT entries at build time with the gen_gdt.py script. We can make TLS depend on CONFIG_GDT_DYNAMIC. It ends up in the data section and we should be able to update it at runtime on context switch. There is no SMP support for 32-bit so no need to worry about that.

64-bit this instead uses the `%fs` segment. The GDT is hard-coded in locore.S, and you can see where we create the TSS descriptors for each CPU, probably just need to augment each one with TLS descriptors (and maybe use a macro instead of repeating ourselves up to 4 times). Same deal should be fairly straightforward to update on context switch, probably where we hook into other C code at the top of `__resume`.

keith-packard, 2020-09-25:
We can just re-use the DS descriptor for the TLS segment register and then adjust the base address using the MSR (for 32-bit mode) or maybe the new wrfsbase instruction in 64-bit mode.  We could create a separate GDT entry in 32-bit mode to set the length field to match the TLS size if desired; that would catch stray references through gs. There's no need to mess with the GDT itself at runtime; the only thing we want to adjust is the base address.

nashif, 2020-12-01:
Already in the tree.

closed 2020-12-01 by nashif