zephyrproject-rtos/zephyr #22204: CONFIG_BT_DEBUG_LOG vs atomic_ operations

novalisek, 2020-01-26:
**Describe the bug**
Hi, I ty to use atomic_clear in my app, but it is crashing. I found out that it has something to do with  CONFIG_BT_DEBUG_LOG

**To Reproduce**
Steps to reproduce the behavior:

Add this to prj.conf
```
CONFIG_BT=y
CONFIG_BT_DEBUG_LOG=y
```
and those lines to main.c
```
atomic_t test = ATOMIC_INIT(0);
void main(void)
{
	atomic_val_t last = atomic_clear(test);

```

then I get

> *** Booting Zephyr OS build zephyr-v2.1.0-581-g4e135d76a3c2  ***
> [00:00:00.005,126] <err> os: ***** MPU FAULT *****
> [00:00:00.005,126] <err> os:   Data Access Violation
> [00:00:00.005,157] <err> os:   MMFAR Address: 0x0
> [00:00:00.005,157] <err> os: r0/a1:  0xfffffff5  r1/a2:  0x20001b50  r2/a3:  0x00000000
> [00:00:00.005,157] <err> os: r3/a4:  0x00000000 r12/ip:  0xe000ed90 r14/lr:  0x00007681
> [00:00:00.005,157] <err> os:  xpsr:  0x61000000
> [00:00:00.005,157] <err> os: Faulting instruction address (r15/pc): 0x0000058a
> [00:00:00.005,157] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
> [00:00:00.005,187] <err> os: Current thread: 0x20000c78 (unknown)
> [00:00:00.069,976] <err> os: Halting system
> 

When I disable CONFIG_BT_DEBUG_LOG, there is no fault

**Expected behavior**
There should be no MPU FAULT

**Environment (please complete the following information):**
 - board:  nrf52840_pca10040
 - OS: W7
 - Toolchain: GNU ARM Embedded 9-2019-q4
 - Commit SHA or Version used: 4e135d76a3c222f208a8e0a3dce93dde55671a4e

jhedberg, 2020-01-26:
Your code:

```
atomic_t test = ATOMIC_INIT(0);
void main(void)
{
	atomic_val_t last = atomic_clear(test);
```

`atomic_clear()` takes a pointer to atomic - you seem to be passing by value instead of by reference, which would explain strange behaviour (probably nothing to do with logging)

novalisek, 2020-01-26:
Oh sorry,
that was a stupid error, I guess I was overwriting wrong memory. I should better watch the warnings in Eclipse :) Sorry once again for false alarm.

closed 2020-01-26 by novalisek