zephyrproject-rtos/zephyr #22207: Bluetooth ：Mesh：Provison init should after proxy

moringe, 2020-01-26:
https://github.com/zephyrproject-rtos/zephyr/blob/bce335458bbf9fbeeb82c16073773a46a3d4533e/subsys/bluetooth/mesh/main.c#L292-L328

Provision use pb-gatt use buffer from proxy client, however proxy client init after prov init.

jhedberg, 2020-01-27:
You're right, this looks like a real bug - I'm surprised no one has noticed since it results in the provisioning buffer having `buf->data` pointing to NULL. The fix should be as simple as reordering the calls in `bt_mesh_init()` i.e. so that `bt_mesh_proxy_init()` gets called before `bt_mesh_prov_init()`.

jhedberg, 2020-01-27:
@moringe can you confirm that the following would solve the issue:

```
--- a/subsys/bluetooth/mesh/main.c
+++ b/subsys/bluetooth/mesh/main.c
@@ -304,6 +304,10 @@ int bt_mesh_init(const struct bt_mesh_prov *prov,
                return err;
        }
 
+       if (IS_ENABLED(CONFIG_BT_MESH_PROXY)) {
+               bt_mesh_proxy_init();
+       }
+
        if (IS_ENABLED(CONFIG_BT_MESH_PROV)) {
                err = bt_mesh_prov_init(prov);
                if (err) {
@@ -316,10 +320,6 @@ int bt_mesh_init(const struct bt_mesh_prov *prov,
        bt_mesh_beacon_init();
        bt_mesh_adv_init();
 
-       if (IS_ENABLED(CONFIG_BT_MESH_PROXY)) {
-               bt_mesh_proxy_init();
-       }
-
        if (IS_ENABLED(CONFIG_BT_SETTINGS)) {
                bt_mesh_settings_init();
        }
```

PR #22230 (2020-01-29): Bluetooth: Mesh: Fix provisioning buffer initialization
When PB-GATT support has been enabled the provisioning code "borrows"
the buffer from the proxy code. However, the way that initialization
was happening the proxy buffers were initialized only after
provisioning initialization, resulting in a corrupted buffer with
buf->data pointing to NULL. Reorder the initialization calls so that
proxy is done first and provisioning only after it.

Fixes #22207
commit 637582df (2020-01-29): Bluetooth: Mesh: Fix provisioning buffer initialization

When PB-GATT support has been enabled the provisioning code "borrows"
the buffer from the proxy code. However, the way that initialization
was happening the proxy buffers were initialized only after
provisioning initialization, resulting in a corrupted buffer with
buf->data pointing to NULL. Reorder the initialization calls so that
proxy is done first and provisioning only after it.

Fixes #22207
closed 2020-01-29 by jhedberg