zephyrproject-rtos/zephyr #22208: gpio: clean up debounce configuration

pabigot, 2020-01-26:
Currently any GPIO peripheral that supports a debounce capability must respect the `GPIO_INT_DEBOUNCE` flag.  As has been raised multiple times (most recently https://github.com/zephyrproject-rtos/zephyr/pull/22139#discussion_r370878795) there are problems with this:
* The flag is named in a way that suggests it should be provided to `gpio_pin_interrupt_configure()`, but instead it must be provided to `gpio_pin_configure()`;
* The flag is not compatible with the way devicetree specifies debounce, which is by a time constant (`gpio-keys` uses the property `debounce-interval` measuring milliseconds).

This incongruity could not be resolved during the GPIO API rework.  See additional commentary at #19649.

pabigot, 2020-01-26:
@carlescufi fyi

koffes, 2020-05-13:
Let me know if you want this as a separate ticket, but it falls reasonably well under the current title.

From the doc: "Drivers that do not support a debounce feature should ignore this flag rather than rejecting the configuration with -ENOTSUP." I can imagine many reasons why this has been done, however, this solution leads developers into a trap. They set they set the `GPIO_INT_DEBOUNCE` flag, get no error and believe that debounce is active even if the platform does not support it. 

Suggest that either we go back to rejecting the configuration, or, if that is not possible, print a warning.

billy-king, 2021-08-19:
May I ask if there is any follow-up to this discussion?
When I develop the driver for our GPIO device with zephyr I found that when I use the API "gpio_pin_configure" to enable the interrupt debouncing, there is no parameter to assign the debounce interval. 
How can I determine the debounce interval for each GPIO interrupt? 
Can I only use the DTS property or Kconfig to set debounce interval at compiler time?