zephyrproject-rtos/zephyr #22210: Bluetooth -  bt_gatt_get_value_attr_by_uuid

novalisek, 2020-01-26:
**Is your enhancement proposal related to a problem? Please describe.**
I am trying to use indications/notifications and when you have multiple characteristics in services, it is sometimes hard (you have to take care of it everytime you change charactericist order or add/remove them) to get correct index to get correct attribute from service.attrs[], which is passed as arg to bt_gatt_notify or to bt_gatt_indicate/notifi_params.attr

**Describe the solution you'd like**
I would like to have function like  
`const struct bt_gatt_attr *bt_gatt_get_value_attr_by_uuid(const struct bt_uuid *uuid);`
which would return correct pointer to value attribute by characteristic uuid

**Describe alternatives you've considered**
helper macros like BT_GATT_CHARACTERISTIC() could also have one more parameter "name" and could create pointers to service.attrs[], and you could use this named pointer later.

or may be  functions like
```
bt_gatt_notify_by_uuid(struct bt_conn *conn, const struct bt_gatt_uuid *uuid,  const void *data, u16_t len);
bt_gatt_indicate_by_uuid(struct bt_conn *conn, const struct bt_gatt_uuid *uuid,  const void *data, u16_t len);
```
but they would have to search for correct attributes in every call ... using  bt_gatt_get_value_attr_by_uuid  you could store the pointer one time and then just use it ...

Vudentz, 2020-02-03:
@novalisek There does exist a shell command that does exactly that, so yes it is possible to do it already, and both bt_gatt_notify and bt_gatt_indicate can have a UUID set to do just what you are proposing: 

https://github.com/zephyrproject-rtos/zephyr/blob/master/include/bluetooth/gatt.h#L828
https://github.com/zephyrproject-rtos/zephyr/blob/master/include/bluetooth/gatt.h#L911

novalisek, 2020-02-03:
@Vudentz what is that command? you can pass UUID to notify/indicate params, but you still have to pass "attr" parameter https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/bluetooth/host/gatt.c#L1857  and that's what I don't want. I would like to have some function, that will find the "attr" param for me by UUID.

joerchan, 2020-02-04:
```
static u8_t match_uuid(const struct bt_gatt_attr *attr, void *user_data)
{
	const struct bt_gatt_attr **found = user_data;

	*found = attr;

	return BT_GATT_ITER_STOP;
}

struct bt_gatt_attr *attr;
bt_gatt_foreach_attr_type(handle, 0xffff, uuid, NULL, 1, match_uuid, &attr);
```
@novalisek  You could use `bt_gatt_foreach_attr_type` to find the attribute you are looking for, see example, you can limit the search using the handles to limit it to the service for example.

@Vudentz I think it is a good point made by @novalisek. We should allow attr to be NULL when UUID is passed instead.

novalisek, 2020-02-04:
@joerchan thanks. I will try that
I still think that this function UUID2attr could be useful, becase if you only allow attr to be NULL when UUID is set, then every call of notify/indicate will have to search for attr again and again ... With this function, you could call it once, save the pointer and just pass it to notify/indicate. I know that it may be not too much instructions to search it everytime, but I like to optimize ... :)

caco3, 2021-05-03:
I also believe such a function would be useful.
I have notifications in a primary and a secondary service and was unable to get the indexes with above example.
(ended up to find it with try & error)

joerchan, 2021-05-11:
Re-opening due to feedback.
1. Provide helper function get attribute from UUID.
2. Allow attr=NULL when passing UUID.

novalisek, 2021-06-16:
@joerchan ok, thank you, both commits looks good to me.. It should be no problem to set the param->attr to  array defined by BT_GATT_SERVICE_DEFINE.
I will be playing with BT again in a week or two, so I will use this new functionality.  I think I will be using bt_gatt_find_by_uuid to store the pointers first and then use them, so that they are not searched in every notify/indicate, it may save few cycles :)

PR #36330 (2021-06-22): Bluetooth: host: Add find by UUID helper function for application.
Allow to pass attribute as NULL pointer when using notify or indicate by
UUID. This will use the entire handle value range to search for an
attribute with a matching UUID.
Document optional parameters, and clarify attr and uuid usage in the
variable declaration in the struct for clarification.

Expose a helper function to the application that searches the local
database for the given attribute from its UUID.
Provide arguments to limit the search that matches the service
declaration to make it easy to limit the search to a speci [... cut]
closed 2021-06-22 by nashif