zephyrproject-rtos/zephyr #22222: Enabling OpenThread SLAAC

braiden, 2020-01-27:
**Is your enhancement proposal related to a problem? Please describe.**
My OpenThread device is not configuring its addresses via SLAAC. It appears that SLAAC is not enabled in zephyr's build of openthread.

**Describe the solution you'd like**
A KConfig? or, just enable by default?

**Describe alternatives you've considered**
`set(OT_SLAAC ON)` in my project's own `CMakeLists.txt` seems to work, but that seems like a hack and now how i'm supposed to enable the feature(?).

**Additional context**

rlubos, 2020-01-27:
I'd keep it configurable, just as some other OpenThread features already are. `OPENTHREAD_SLAAC` on the Zephyr side plus the following entry in the OpenThread CMake file should do the job:
```
if(CONFIG_OPENTHREAD_SLAAC)
  set(OT_SLAAC ON CACHE BOOL "...")
endif()
```

rlubos, 2020-05-05:
SLAAC was added in https://github.com/zephyrproject-rtos/openthread/pull/6

closed 2020-05-05 by rlubos