zephyrproject-rtos/zephyr #22245: STM32G431: Wrong SystemCoreClock variable

martinjaeger, 2020-01-27:
**Description**
Because of a bug in STM32Cube LL, the `SystemCoreClock` variable is set wrong if the actual system frequency is >80 MHz.

More details are described in a related (but already closed) PR:
https://github.com/zephyrproject-rtos/hal_stm32/pull/34#issuecomment-578881473

I raised this issue to track the bug, as I can't raise an issue in the original HAL repository.

**Impact**
Boards with STM32G431 are hardly useable with high clock frequencies, as e.g. UART baud rates are get configured wrong.
Preliminary workaround could be to reduce the system frequency to <80 MHz.

ulfalizer, 2020-01-28:
Semi-unrelated, but could we enable issues on module repos? @nashif

rosterloh, 2020-01-31:
https://github.com/zephyrproject-rtos/hal_stm32/pull/36

closed 2020-02-26 by MaureenHelm