zephyrproject-rtos/zephyr #22259: default value 80 on BT_ACL_RX_COUNT clamped to 64

mniestroj, 2020-01-28:
**Describe the bug**
There is a warning in at least one sample:
```
warning: default value 80 on BT_ACL_RX_COUNT (defined at subsys/bluetooth/host/Kconfig:244) clamped to 64 due to being outside the active range ([1, 64])
```
This warning turns into an error when compiled under sanitycheck.

**To Reproduce**
Steps to reproduce the behavior:
1. Apply #21206, which solves #22258
2. /scripts/sanitycheck -p qemu_x86 -T samples/net/sockets/echo_client/ -i
3. See error:

```
$ ./project/zephyr/scripts/sanitycheck -p qemu_x86 -T project/zephyr/samples/net/sockets/echo_client/ -iRenaming output directory to /zephyr/sanity-out.3
INFO    - JOBS: 8
INFO    - Building initial testcase list...
INFO    - 3 test configurations selected, 4 configurations discarded due to filters.
INFO    - Adding tasks to the queue...
ERROR   - Cmake build failure: /zephyr/project/zephyr/samples/net/sockets/echo_client for qemu_x86

ERROR   - qemu_x86                  samples/net/sockets/echo_client/sample.net.sockets.echo_client.bt FAILED: Cmake build failure
INFO    - /zephyr/sanity-out/qemu_x86/samples/net/sockets/echo_client/sample.net.sockets.echo_client.bt/build.log
ERROR   - -- Zephyr version: 2.1.99
-- Found PythonInterp: /usr/bin/python (found suitable version "3.8.1", minimum required is "3.6") 
-- Selected BOARD qemu_x86
-- Found west: /usr/bin/west (found suitable version "0.6.3", minimum required is "0.6.0")
-- Loading /zephyr/project/zephyr/boards/x86/qemu_x86/qemu_x86.dts as base
Devicetree configuration written to /zephyr/sanity-out/qemu_x86/samples/net/sockets/echo_client/sample.net.sockets.echo_client.bt/zephyr/include/generated/devicetree.conf
Parsing /zephyr/project/zephyr/samples/net/sockets/echo_client/Kconfig
Loaded configuration '/zephyr/project/zephyr/boards/x86/qemu_x86/qemu_x86_defconfig'
Merged configuration '/zephyr/project/zephyr/samples/net/sockets/echo_client/prj.conf'
Merged configuration '/zephyr/project/zephyr/samples/net/sockets/echo_client/boards/qemu_x86.conf'
Merged configuration 'overlay-bt.conf'

warning: default value 80 on BT_ACL_RX_COUNT (defined at subsys/bluetooth/host/Kconfig:244) clamped to 64 due to being outside the active range ([1, 64])

error: Aborting due to non-whitelisted Kconfig warning 'warning: default value 80 on BT_ACL_RX_COUNT
(defined at subsys/bluetooth/host/Kconfig:244) clamped to 64 due to being outside the active range
([1, 64])'. If this warning doesn't point to an actual problem, you can add it to the whitelist at
the top of /zephyr/project/zephyr/scripts/kconfig/kconfig.py.

CMake Error at /zephyr/project/zephyr/cmake/kconfig.cmake:214 (message):
  command failed with return code: 1
Call Stack (most recent call first):
  /zephyr/project/zephyr/cmake/app/boilerplate.cmake:461 (include)
  CMakeLists.txt:5 (include)

-- Configuring incomplete, errors occurred!

INFO    - /zephyr/sanity-out/qemu_x86/samples/net/sockets/echo_client/sample.net.sockets.echo_client.bt/build.log
INFO    - Total complete:    3/   3  100%  skipped:    0, failed:    1
INFO    - 2 of 3 tests passed (66.67%), 1 failed, 0 skipped with 0 warnings in 15.22 seconds
INFO    - In total 7 test cases were executed on 1 out of total 220 platforms (0.45%)
```

**Expected behavior**
Sanitycheck test should pass.

**Impact**
Sanitycheck tests cannot pass anymore after applying #21206.

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK
 - Commit SHA or Version used: c793c0edda931ab95644ee3bf977600a88266260 with #21206 applied.

aescolar, 2020-01-28:
CC @jhedberg

joerchan, 2020-01-28:
This follows from:
CONFIG_NET_BUF_RX_COUNT=80

config BT_ACL_RX_COUNT
	int "Number of incoming ACL data buffers"
	default BT_CTLR_RX_BUFFERS if BT_CTLR
	default NET_BUF_RX_COUNT if NET_L2_BT
	default 6
	range 1 64
	help

NET_BUF_RX_COUNT does not have a range, which leads us to this situation.

@rlubos Does this need to be 80?

rlubos, 2020-01-28:
The buffer count was finetuned so that the sample can run smoothly in full configuration (both IPv4/IPv6 and UPD/TCP). So yes, the default configuration should stay as it is.

I'm not sure why do you need to limit the maximum number for buffers for BLE, but if that's the case you could do it in the `overlay-bt.conf` file. Should still work fine as for BLE IPSP we run only IPv6.

PR #22381 (2020-01-31): samples: net: set CONFIG_NET_BUF_RX_COUNT to 64
This count is assigned to CONFIG_BT_ACL_RX_COUNT which expects a number
in the range of 1-64, otherwise kconfig fails.

Fixes #22259
commit e3acaa10 (2020-01-31): samples: net: set CONFIG_NET_BUF_RX_COUNT to 64

This count is assigned to CONFIG_BT_ACL_RX_COUNT which expects a number
in the range of 1-64, otherwise kconfig fails.

Fixes #22259
closed 2020-01-31 by nashif