zephyrproject-rtos/zephyr #22264: drivers: serial: nrf_uart & nrf_uarte infinite hang

JordanYates, 2020-01-28:
**Describe the bug**
Both drivers fail to properly mediate access to the underlying hardware when used in poll mode.
Both drivers (uart_nrfx_uart.c and uart_nrfx_uarte.c) in their poll_out implementations, immediately start modifying hardware registers, then busy-wait on a hardware register.

Once one thread starts sending and blocks, a second thread will go through the same motions and block on the same register. Given that they clear the register on success, only one will ever be unblocked, leaving the other stuck.

The simplest scenario in which this happens is when using the logging subsystem and printk in a thread.  When the TX's collide, the lower priority thread hangs.

**To Reproduce**`

The simplest way to reproduce this issue is to smash the logger with log messages, and try and use printk. The easiest application for that is `samples/bluetooth/scan_adv`
proj.conf additions:
```
CONFIG_BT_LOG_LEVEL_DBG=y
```
main.c additions:
```
/* In the infinite loop */
printk("TEST\n");
```
Compile and run:
```
west build -b nrf52840_pca10056 zephyr/samples/bluetooth/scan_adv
west flash
```
Open your serial program of choice, and the bluetooth log messages will typically stop the first time "TEST" prints.

**Expected behavior**
The logging thread should not infinitely hang.

**Impact**
Major problem, UART subsystem should not break on timing collisions.

**Screenshots or console output**
Example output of the time at which the logger hangs
```
[00:00:00.266,113] <dbg> bt_hci_core.bt_recv: buf 0x20003288 len 44
[00:00:00.266,113] <dbg> bt_hci_core.hci_event: event 0x3e
[00:00:00.266,113] <dbg> bt_hci_core.hci_le_meta_event: subevent 0x02
[00:00:00.266,143] <dbg> bt_hci_core.le_adv_report: Adv number of reports 1EST
TEST
TEST
TEST
TEST
TEST
```

**Environment (please complete the following information):**
 - Commit SHA: 5af54075703c248fae2bf69a50909bf2e3cc977e

Mierunski, 2020-01-30:
I haven't noticed the bug in UARTE. Proposed a PR with fix for UART.
Note that using single UART instance from multiple contexts can result in fragmented/altered data. If you want such functionality, check tty API

JordanYates, 2020-01-30:
I will try and replicate the UARTE version on another zephyr installation on another computer. I understand the fragmented data problem, given the current polling implementation of sending one character per call. However from my understanding the async api should never result in fragmented data.

JordanYates, 2020-01-31:
Performed a clean install of zephyr from the v2.1 branch on a different computer.
After following the UARTE variant instructions, problem occurs 100% of the time on the first print.
Perhaps your environment does not have enough bluetooth packets to fill the logger?
Do you get dropped packets on the log?

Mierunski, 2020-01-31:
I managed to reproduce the error (idk why it didn't reproduce first time), proposed a fix in PR.

PR #22355 (2020-03-13): drivers: nrf: Fix UART TX hanging when using both polling and async API.
Fix UART TX hanging when using both polling and async API.

fixes #22264
closed 2020-03-13 by ioannisg