zephyrproject-rtos/zephyr #22268: timer not working when duration is too high

ifillonneau, 2020-01-28:
**Describe the bug**
Starting a timer with huge time make the timer trigger without waiting.
Ex: k_timer_start(&timer, K_HOURS(19), 0); does not work at all whereas k_timer_start(&timer, K_HOURS(18), 0); is working. The problem seems to come from conversion from duration to tick witch is limited to signed 32 bits. as soon as the conversion in tick is greater than 0x7FFFFFFF, the timer is not working at all.

**To Reproduce**
```c

#include <zephyr.h>
#include <kernel.h>

static void handler(struct k_timer *dummy)
{
	printk("handler called\n");
}

K_TIMER_DEFINE(timer, handler, NULL);

void main(void)
{
	printk("test Timer !\n");
#ifdef NOT_WORKING
	k_timer_start(&timer, K_HOURS(19), 0); // handler is called immediately
#else
	k_timer_start(&timer, K_HOURS(18), 0); // handler is called after 18 hours
#endif
}
```

Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=nrf52_pca10040 path_to_above_main
3. make

**Expected behavior**
timer to trigger after 19 hours

**Impact**
showstopper

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK
 - Commit SHA or Version used: 10fb64f4dc12ef946

SebastianBoe, 2020-01-28:
Duplicate of https://github.com/zephyrproject-rtos/zephyr/issues/19075 ?

ifillonneau, 2020-01-28:
@SebastianBoe , yes, this is the same behavior, and the same root cause

jhedberg, 2020-01-28:
Closing since #19075 seems to be a duplicate

closed 2020-01-28 by jhedberg