zephyrproject-rtos/zephyr #22272: aggregated devicetree source file needs to be restored to build directory

pabigot, 2020-01-28:
#21785 included what became 8d8b06978c577bff24006aa3b3c875d9dae820d0 which removes the generated aggregate devicetree source file.

This file is critical for diagnosing problems in devicetree content and syntax.  That it's not used by the tooling anymore does not mean it's not used.  There is no other reasonable way to confirm that specific overlays are being made available to the generation scripts.

We do seem to have `board.dts.pre.d` and `board.dts.pre.tmp` still available, but there's no promise that they have meaningful content, unlike `.dts_compile` which I could rely on to be an accurate representation of the devicetree content used when generating code.

ulfalizer, 2020-01-28:
`dtlib.py` can dump the devicetree too (while preserving more info than the C tools, like labels). I could save the output from that somewhere.

Will look tomorrow.

pabigot, 2020-01-28:
That would be helpful, thank you.  Right now I'm down a rabbit hole trying to figure out why a node that's present in the devicetree source with status `okay` is listed in the dependency tree but has no binding macros generated for it.

pabigot, 2020-01-28:
The answer is that, among other things, `compatible` was missing.  Could be useful to diagnose that.

ulfalizer, 2020-01-28:
This should work as a quick hack:

```diff
diff --git a/scripts/dts/edtlib.py b/scripts/dts/edtlib.py
index 24ff5ff77d..dfb7d7e78b 100644
--- a/scripts/dts/edtlib.py
+++ b/scripts/dts/edtlib.py
@@ -124,6 +124,7 @@ class EDT:
         self.bindings_dirs = bindings_dirs
 
         self._dt = DT(dts)
+        print(self._dt)
         _check_dt(self._dt)
 
         self._init_compat2binding(bindings_dirs)
```

> The answer is that, among other things, `compatible` was missing. Could be useful to diagnose that.

Yeah, useful to have available. Will add.

Wonky sleep schedule. Off to bed now.

galak, 2020-01-28:
I think PR #22282 is an attempt to address this issue.

mbolivar, 2020-01-28:
I wasn't aware of the bug being filed, but I agree it is one.

@ulfalizer you should also be aware (if you are not already) that 4e85006ba455f7434ded5cf933ac85d2b1b01ff6 still mentions BOARD.dts_compiled in both zephyr_dt_inputs_outputs.png and zephyr_dt_inputs_outputs.svg, so the current docs continue to refer to this file.

mbolivar, 2020-01-28:
@jhedberg I think this is at least medium priority personally.

pabigot, 2020-01-28:
I must have been tuned out during the discussion of this, but it absolutely should be fixed before 2.2 so I'd support a reprioritization.

ulfalizer, 2020-01-29:
> @ulfalizer you should also be aware (if you are not already) that [4e85006](https://github.com/zephyrproject-rtos/zephyr/commit/4e85006ba455f7434ded5cf933ac85d2b1b01ff6) still mentions BOARD.dts_compiled in both zephyr_dt_inputs_outputs.png and zephyr_dt_inputs_outputs.svg, so the current docs continue to refer to this file.

Thought I had fixed those up in commit dbed9027ba ("doc: build overview: Improve diagram, fix errors/stale info, misc."). I'll check if I missed any spots.

jhedberg, 2020-01-29:
> @jhedberg I think this is at least medium priority personally.

Updated to medium.

ulfalizer, 2020-01-29:
Fix: https://github.com/zephyrproject-rtos/zephyr/pull/22295

andrewboie, 2020-01-30:
**Why is this a medium?**
This does not represent an issue on the device/kernel itself, just a validation layer for devicetree contents/syntax!
If you can show that this directly causes the device to malfunction, keep it at medium, but if this is a build system validation thing please don't add this to our ballooning P2 count, which is full of bugs where the device is unstable and crashing

closed 2020-01-30 by galak

mbolivar, 2020-01-30:
> If you can show that this directly causes the device to malfunction, keep it at medium, but if this is a build system validation thing please don't add this to our ballooning P2 count, which is full of bugs where the device is unstable and crashing

Where is this documented as a criterion for bug severity? I don't agree that medium and up should be limited to just the firmware.

commit edfe5a65 (2020-01-30): dts: gen_defines.py: Save final devicetree to zephyr/zephyr.dts

Add a '--dts-out <file>' flag to gen_defines.py that saves the final
devicetree to <file>, in DTS format, using the new EDT.dts_source
attribute.

Handy to have available as a debugging aid. Unused otherwise.

Also write a dummy <BOARD>.dts_compiled file that tells people to look
at zephyr.dts, for people that might be used to that file. It was
removed in commit 8d8b06978c ("dts: Remove generation of
<BOARD>.dts_compiled").

Fixes: #22272