zephyrproject-rtos/zephyr #22290: ARC crashes due to concurrent system calls

andrewboie, 2020-01-28:
**Describe the bug**
I implemented a system call torture test and observed some crashes on nsim_sem and nsim_em (and their mpu_stack_guard variants):

```
Running syscall torture test with 4 threads on 1 cpu(s)
E: ***** Exception vector: 0x6, cause code: 0x1, parameter 0x6
E: Address 0x800028e0
E: EV_ProtV
E: Memory read protection violation (unknown)
E: >>> ZEPHYR FATAL ERROR 2: Stack overflow on CPU 0
E: Current thread: 0x80000380 (unknown)
E: Halting system
Console now belongs to simulator.
[100%] Built target run
```

```
starting test - test_syscall_torture
Running syscall torture test with 4 threads on 1 cpu(s)
E: ***** Exception vector: 0x3, cause code: 0x0, parameter 0x0
E: Address 0x2b0c
E: EV_MachineCheck
E: double fault
E: Faulting instruction address = 0x2b0c
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x80000a90 (unknown)
E: Halting system
Console now belongs to simulator.
```

**To Reproduce**
Steps to reproduce the behavior:
1. Pull this patch which introduces the test: https://github.com/zephyrproject-rtos/zephyr/pull/22288
1. Run tests/kernel/mem_protect/syscalls. `sanitycheck -a arc -T tests/kernel/mem_protect/syscalls` ought to do it. `nsimdrv` needs to be in PATH.
1. Observe the tests failing on all ARC nsim platforms, either due to protection violations or double faults.

**Expected behavior**
Passing test result.

**Impact**
System calls are unstable on ARC which could be a showstopper for some people.

**Additional context**
Might have a similar root cause as #22275 but I haven't looked at the ARC MPU code yet.

andrewboie, 2020-01-28:
@vonhust @abrodkin @ruuddw FYI

ruuddw, 2020-01-29:
Thanks, good to have the torture test.
Root cause could be similar for nsim_sem. MPU v3 has a index control register like arm. nsim_em uses MPU v2 which has separate config registers per-index, so probably some other failure.

vonhust, 2020-02-01:
@andrewboie  @ruuddw 

ARC's mpu implementation has similar issue of ARM mpu, also found some new issue, i am working on a fix.

andrewboie, 2020-02-06:
@vonhust I think we can close this with your PR merged, do you agree?

andrewboie, 2020-02-06:
Closing this, but some new regressions have popped up in semaphore tests, I filed #22559

closed 2020-02-06 by andrewboie